FreeCAD/FreeCAD #12658: [Problem] Addon Manager doesn't show important properties of an addon

80avin, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Addon Manager doesn't show important properties of an addon, like
* git url
* created/updated date
* author
* version
* Github stars (on description page).  I see sorting by Github stars is added recently, but it also has few issues like
   * Sort option is not persistent.
   * Has a glitch, which can be seen by
      * Open addon manager
      * Before the loading completes (takes 2-3 seconds or more), change the sort field
      * Note that when loading completes, sort field would change to Alphabetical.
* Category
* Size

Few information like author, stars is shown in the list view but not on Addon's info page.

While pointing these issues, I would suggest a better revamp of the Addon Manager UI and probably internals to take inspiration from some popular & similar solutions present in other apps. Like VSCode's Extension search & description page, or Arduino library manager, or some good Software Center.

Further, if the problem & proposal seems acceptable, I can also work to implement a solution.

### Full version info

```shell
OS: Ubuntu Core 22 (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36181 (Git) Snap 940
Build type: Release
Branch: main
Hash: f88f48f64f223085f97f0f2945298f9023d09693
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/India (en_IN)
Installed mods: 
  * CfdOF 1.25.2
  * Render 2024.1.23
  * woodworking 0.21.2.33771
  * Assembly4 0.50.7
  * Assembly3 0.12.2
```
```

### Subproject(s) affected?

None

### Anything else?

Attaching few screenshots for quick reference.

|![image](https://github.com/FreeCAD/FreeCAD/assets/19588421/b2cba92b-40ab-425f-b1f6-8b9521bb7ca6)|![image](https://github.com/FreeCAD/FreeCAD/assets/19588421/ec5e1547-13f1-4dc8-b590-7998725139f5)|
|:---:|:---:|
|Addon List|Addon List (compact)|
|![image](https://github.com/FreeCAD/FreeCAD/assets/19588421/686b9966-284a-42c2-bff5-85d5f7ad95bb)|![image](https://github.com/FreeCAD/FreeCAD/assets/19588421/7e4830ba-bc92-4883-a1b9-498f9016ff13)|
|Addon Description (top)|Addon Description (bottom)|

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

80avin, 2024-02-28:
###### Copy pasting the image table from Issue description
Attaching few screenshots for quick reference.

|[image]|[image]|
|:---:|:---:|
|Addon List|Addon List (compact)|
|[image]|[image]|
|Addon Description (top)|Addon Description (bottom)|

maxwxyz, 2024-02-28:
@chennes

chennes, 2024-02-28:
Thanks, @maxwxyz -- this isn't really one issue, it's at least three. Can you please break it up so that the bugs are individual issues, and the feature request(s) at least one other? Thanks!