FreeCAD/FreeCAD #12659: [Problem] Tech Draw Editing over tolerance format specifier when equal tolerance is selected have issues

beischer, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Double click a dimension to get to the Dimensions panel.
2. Equal Tolerance checkbox checked
3. Try to change the number in the "Edit the Over Tolerance Format Specifier" by deleting the number. That's when the bug happen. The cursor is automatically moved to the end of the string and you have to manually go back to the correct place with your arrow keys or your mouse to enter the number of decimals you want. 

Note that this does not happen when Equal Tolerance is not checked.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git)
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/Sweden (en_SE)
Installed mods: 
  * Assembly4 0.50.6
  * fasteners 0.4.71
  * Help 1.0.3
  * InventorLoader 1.4.0
  * kicadStepUpMod 10.21.9
  * Manipulator 1.5.0
```
```

### Subproject(s) affected?

None

### Anything else?

I discussed this issue on the forum some time ago and think that this was solved, but somehow it is back again (or my memory fails me).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

beischer, 2024-02-28:
@WandererFan I missed to set this to subproject Techdraw when creating the issue. Don't know if that is needed for you to see it.

WandererFan, 2024-03-07:
It seems to me that I'm getting fewer notifications about issues in my inbox.  I don't think I've changed any settings.

WandererFan, 2024-03-07:
OK, I see it now.  Deleting the number in the over specifier triggers a change to under specifier (to a specifier with no number), which in turn triggers a change to over specifier (with the no number specifier) since they are supposed to be equal.  

Need to change how the 2 specifiers are kept in sync.

WandererFan, 2024-03-17:
This is fixed by d7948c5554

closed 2024-03-17 by WandererFan