FreeCAD/FreeCAD #12669: [Problem] `Sketcher_External` doesn’t work well with surfaces

pierreporte, 2024-02-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

`Sketcher_External` projects the selected geometry in the current sketch. It doesn’t work if the selected geometry is non-planar or parallel to the sketch plane. The following errors are generated:

> Failed to add external geometry  
> Non-planar faces are not yet supported for external geometry of sketches

> Failed to add external geometry  
> Selected external reference plane must be normal to sketch plane

The expected result is the outline of the surface projected on the sketch plane.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-02:
@pierreporte please remember to always post the full [About](https://wiki.freecad.org/About) info

pierreporte, 2024-03-17:
@maxwxyz the issues are for two different things. Not sure to understand your second sentence.

Both issues could be integrated into a new one about deprecating `Sketcher_External` in favor of two new commands, `Sketcher_ProjectionExternal` and `Sketcher_IntersectionExternal`, that would be fixed and improved.

maxwxyz, 2024-03-17:
I thought the projected geometry is the same as intersecting one when talking about surfaces. You select a surface as external and would like the projected surface at the position of the sketch plane, therefore you must extend that surface and intersect it with the sketch plane. Or do you have a different example?

pierreporte, 2024-03-17:
I see now.

It is not necessary to extend a surface to project it because, in this case, you want the outline. It’s like projecting the edges manually but more generic. For example, consider a cube that don’t intersect the sketch plane. If you project one of its faces, it must draw a parallelogram in the sketch. If you wanted an intersection of this face, it shouldn’t work, precisely because there is no intersection with the sketch plane. On the other hand, if the same face intersects the sketch plane, the intersection would be a segment (the actual portion of the face in the sketch plane) but the projection is still a parallelogram. Same thing for non-planar surfaces.

pierreporte, 2024-03-17:
We could discuss about extending external geometry for intersection but in #12670.

maxwxyz, 2024-03-17:
so like a TechDraw view is projected, got it. I cannot think of a quick use case but could be useful. I think it's complicated to link 3D and 3D geometry, especially when the underlying geometry changes or the sketch plane is moved or rotated.
Issues to link between 2D/3D were also brought up here: https://github.com/FreeCAD/FreeCAD/issues/12383

pierreporte, 2024-03-17:
I think that the most popular use case is projecting a face parallel to the sketch plane to avoid redrawing the existing edges. Projection of non-planar or non-parallel surfaces is definitely less used but still nice to have.

PaddleStroke, 2024-10-25:
The face part of this issue is actually a duplicate of https://github.com/FreeCAD/FreeCAD/issues/13313
Perhaps this issue here can be kept for 'non planar faces' while the other one is for planar faces that are not normal.

pierreporte, 2024-11-29:
@PaddleStroke is this solved by #17736?

PaddleStroke, 2024-11-29:
Yes

closed 2024-11-29 by pierreporte