FreeCAD/FreeCAD #12670: [Problem] Sketcher: no way to add intersection geometry

pierreporte, 2024-02-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

`Sketcher_External` only projects selected geometry in the sketch plane. The only intersections it can make are with datum geometry, which is very limited. Intersections in sketches are very useful when working with surfaces and wireframe geometry to build-up features.

I would suggest dropping `Sketcher_External` from the UI (not from code) and replace it with `Sketcher_ProjectExternal` as a fixed version (no intersection of datum planes and other reported problems) and `Sketcher_IntersectionExternal` for the intersections.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

TomaszSzt, 2024-03-12:
Just for your information and from an end user point of view who doesn't look into coding details:

[image]

If something like that is what you are looking for, then check the LinkBranch https://github.com/realthunder/FreeCAD

pierreporte, 2024-03-12:
I know that the Link branch can do it but it is not the official FreeCAD. The goal isn’t just the feature but official support.

image.png (view on web)

pierreporte, 2024-03-12:
@realthunder Could you indicate where this is implemented so that the person willing to implement the feature can get inspiration from your work?

pierreporte, 2024-03-17:
Interesting feature: when selected geometry doesn’t intersect with the sketch plane, it could be asked to extend it to compute an intersection. It would be only for easily extendable geometry: planar surfaces (so it isn’t necessary to add a datum plane), straight edges (so no datum line necessary). Surfaces fully normal to the sketch plane could also be extended but it would be equivalent to a projection.

@maxwxyz

realthunder, 2024-03-21:
It is very difficult to isolate this function in my branch. It is interwind with topo naming code I added to Sketcher. I would say wait until topo naming code is merged. External geometry is not reliable without it anyway.

closed 2024-11-29 by WandererFan

AGluk, 2024-12-20:
Hello.

Does anyone know,
Why there is no macOS x64 version of the latest release 2024.10.03 in the LinkBranch?
Only Arm64 version for macOS.

Thanks.

tarman3, 2024-12-20:
Please close this issue and create new in LinkBranch repository.