FreeCAD/FreeCAD #12673: [Problem] Using Transform on a Sub-Shapebinder does nothing and just selects another element.

YosypC, 2024-02-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to move a subshapebinder, but that command does nothing useful.

https://github.com/FreeCAD/FreeCAD/assets/50027752/f3f73a88-7108-4b04-9a64-356dc2bd40fb

If users are not intended to use the Transform tool on SSBs, then it should not be selectable.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36220 (Git)
Build type: Unknown
Branch: main
Hash: 68fd2934cfcdb7b31d235d562412a5c04f55682e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenDark 2023.12.17
  * ThreadProfile 1.89.0
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
I think this is related to https://github.com/FreeCAD/FreeCAD/issues/9435 please confirm.

YosypC, 2024-03-25:
I am not sure actually. It depends by the philosophy, as I don't know FC underlyings. If the command should have an effect on them, so it's a bug. If it's only a UI "leftover" and it should actually not do anything, then it should be removed.

wwmayer, 2024-04-01:
Fixed with https://github.com/FreeCAD/FreeCAD/pull/13261

closed 2024-04-02 by wwmayer