FreeCAD/FreeCAD #12678: [Problem] Support Import/Export of .stpx STEP format

maxwxyz, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to import or export .stpx files into FreeCAD

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2024-03-01 by Roy-043

maxwxyz, 2024-03-01:
@Roy-043 no? I cannot import or open these files. stp**X** not stp**Z**_

maxwxyz, 2024-03-01:
Thanks, happens to me too :) I thought I was doing something wrong not being able to open these files.

luzpaz, 2024-03-01:
I can't locate the `.stpx` format documentation online. Anyone else have better success ?

maxwxyz, 2024-03-01:
I think its from STEP AP242 and has BOM information in it, maybe STEP+XML as https://en.wikipedia.org/wiki/ISO_10303-28 
I currently have no access to the latest version of that standard. 

Related info:
https://en.wikipedia.org/wiki/ISO_10303#Coverage_of_STEP_Application_Protocols_.28AP.29
https://en.wikipedia.org/wiki/ISO_10303-21

dpasukhi, 2024-09-30:
Hello, OCCT plan to implement STPX extensions.
We already working on it And partially implemented internally.
But the point about will it be opensource or commercial not resolved.

_
There are a lot of  STPX documentation, it is "Business" or "Domain model" named.

maxwxyz, 2024-10-06:
@pierreporte I guess this should be on the mechanical board somewhere, to be able to import and export new STEP formats, incl. metadata and annotations

pierreporte, 2024-10-06:
Is this P2 or P3?

maxwxyz, 2024-10-06:
P3 IMO but if it is implemented in OCCT maybe it could be ported over sooner.
@dpasukhi is there a timeline?

dpasukhi, 2024-10-06:
@maxwxyz unfortunately there are no timeline for that activity. That is a "open-time" task for breaks between tasks.
I'm not sure when it will be published. For now only about 20% import/export is ready.

And additionally, I'm not sure about licence. It can be a commercial or open source. That topic on discussion (I'm voting for open source for sure).

maxwxyz, 2024-10-13:
maybe the newest ISO/DIS 10303-242 https://www.iso.org/standard/84300.html can be taken into account, but also information like mates and kinematics for assemblies inside FreeCAD @PaddleStroke 

@pierreporte the STEP standard seems to incorporate even more information now, like version info, dependencies, quality and manufacturing info,... (latest one: https://www.iso.org/standard/84667.html) Is this worth putting on the mechanical list? IMO supporting good import and export for the industry standard format is important for wider adoption of FreeCAD.

pierreporte, 2024-10-13:
Yes it would be good if FreeCAD was fully compliant to newer STEP standard, even though it doesn’t have all the features yet. Do you want to create a new issue? If so I’ll put it in the board.

maxwxyz, 2025-02-23:
@pierreporte finally done

maxwxyz, 2025-05-14:
according to https://dev.opencascade.org/node/88020#comment-26960 it should be available this quarter, @pieterhijma

dpasukhi, 2025-05-14:
STPX is already available for import/export as a commercial component.
The full list:https://occt3d.com/data-exchange-formats/
Soon we start delivery as planned non-commercial binary delivery.