FreeCAD/FreeCAD #12679: [Feature Request] [Part Design] Making of external threads simplified

pok00, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Few months ago I tried freecad instead of other designing softwares and I love it is well made, but I miss a feature for making metric external threads like some other softwares have. Making internal threads in freecad is very simple using hole feature:
[image]
I would also love similar feature for external thread making since everybody uses helix tool where you need to get everything correctly and is quite time consuming. If this feature would get implemented for external threads I could entirely migrate to freecad software and it would be perfect since I can also use it on my linux laptop. I miss this feature a lot especially for 3d printing. The internal threads already 3d prints flawlessly.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
Is this different to https://github.com/FreeCAD/FreeCAD/issues/5567
If not, please comment there.

pok00, 2024-03-01:
I think its different he is asking for cosmetic threads for drawings, but I am asking for functional threads in part design that I can 3d print and is correct pitch and step etc.

Roy-043, 2024-03-01:
Maybe check out the Fasteners Workbench?:
https://wiki.freecad.org/Fasteners_Workbench#Rods,_taps_and_dies

pok00, 2024-03-02:
I tried few thread making plugins including fastener workbench, but I didn't really like the way they work.

mpfusion, 2024-06-20:
Creating external thread with the Fasteners Workbench works, but it requires another workbench and a completely different way to create threads compare to the inner thread using the hole feature.

I think it makes sense to have a tool equivalent to the hole feature, just the other way round. That would be much more intuitive and make for a more consistent UI. The thread would then be part of the Body so there's no need to union/fuse the thread afterwards (as it is with the Fasteners Workbench).

maxwxyz, 2024-08-25:
https://github.com/FreeCAD/FreeCAD/issues/14274 is kind of similar, but not only for threads.

pierreporte, 2024-12-14:
This should be fixed by how the new thread feature has been specified: https://github.com/FreeCAD/FreeCAD/issues/5567#issuecomment-2543142577

closed 2025-03-09 by maxwxyz