FreeCAD/FreeCAD #12680: [Problem] Managing modified files in the session is hard

pierreporte, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working on several parts in a FreeCAD session, closing the program will prompt dialog boxes to ask if the user wants to save the part, one per modified file. This is mostly fine but the users needs to be very careful to answer right to each of the dialog, that all look the same because the file name is buried in the text.

[image]

When working in an assembly, a lot of parts and assemblies can be modified (especially parts and datum made visible or hidden) to get the work done. Oftentimes, only a few parts are going to be saved. However, when closing, it asks the user to save all modified components or not after the same dialog for the assembly container.

[image]

FreeCAD should have a way to make clearer what files are modified and give a choice to save only the ones that needs to be. I propose to replace these dialogs by a new one that lists all modified files in the session, with checkboxes at every line of the list, and a button to select all and deselect all lines. Other programs have a similar save management dialog. They allow to change the file name and location as well.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Anything else?

[image]

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-01:
The lines in the dialog could use the format defined for #12141 for easier use. When working with a PDM, the file in cache may not be useful (it can be a hash or similar). The file name should still be displayed of course.

pierreporte, 2024-05-06:
@PaddleStroke This is going to be useful for working with assembly. What’s your opinion of this dialog?

PaddleStroke, 2024-05-07:
This is a relatively large feature request. So it can't make it to 1.0 feature freeze