FreeCAD/FreeCAD #12681: [Problem] Sketcher: circle and arc OVPs should be able to switch to diameter

pierreporte, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The OVPs for circles and arcs are currently the radius. It should be possible to easily switch to diameter, and back.

For the circle, the diameter should be the default parameter.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
Is this different to: https://github.com/FreeCAD/FreeCAD/issues/12384 ?

maxwxyz, 2024-03-01:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12384

closed 2024-03-01 by maxwxyz

pierreporte, 2024-03-01:
I made sure too search for existing issues. The default part is duplicate but the switching is not. I will edit to keep only this.