FreeCAD/FreeCAD #12682: FreeCAD crashes when inserting windows and doors[Problem]

cpmach, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad crashes when trying to insert a door or a window in a wall.
This just started to happen after the upgrade to version 0.21.2! I even tryed to use the latest upgrade but the problem pressist.
Can you help me? Or should I need to go back to 0.20 version?

### Full version info

```shell
[code]
OS: macOS 11.7.10
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Reporting
  * Behave-Dark-Colors 0.1.1
  * parts_library
  * Render 2023.8.18
  * ArchTextures
  * A2plus 0.4.60n
  * BIM 2021.12.0
  * Curves 0.6.14
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

Arch

### Anything else?

rocess:               freecad [16464]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            org.freecadweb.FreeCAD
Version:               ??? (0.21.2-33771)
Code Type:             X86-64 (Native)
Parent Process:        ??? [16458]
User ID:               501

Date/Time:             2024-03-01 14:49:18.944 +0000
OS Version:            macOS 11.7.10 (20G1427)
Report Version:        12
Anonymous UUID:        3890C32F-F854-CEEB-B8D1-4219E4506269

Sleep/Wake UUID:       5AF3FDF2-0BA5-4FD0-8CB0-AC610C77BBA1

Time Awake Since Boot: 320000 seconds
Time Since Wake:       2100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [16464]
[... 391 lines cut ...]
Stack                            179.7M       29 
VM_ALLOCATE                      432.1M      141 
VM_ALLOCATE (reserved)           256.0M        2         reserved VM address space (unallocated)
__DATA                           146.2M      715 
__DATA_CONST                      25.3M      301 
__DATA_DIRTY                      1691K      206 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       564.4M      355 
__OBJC_RO                         70.3M        1 
__OBJC_RW                         2496K        3 
__TEXT                           755.0M      663 
__UNICODE                          588K        1 
mapped file                      444.0M       61 
shared memory                      752K       15 
===========                     =======  ======= 
TOTAL                              3.7G     2718 
TOTAL, minus reserved VM space     3.0G     2718 

Model: iMac18,3, BootROM 515.0.0.0.0, 4 processors, Quad-Core Intel Core i5, 3,4 GHz, 40 GB, SMC 2.41f2
Graphics: kHW_AMDRadeonPro570Item, Radeon Pro 570, spdisplays_pcie_device, 4 GB

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
Can you provide a demo file or steps on how to reproduce this?
Also if possible, please try if the behavior is fixed in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

cpmach, 2024-03-01:
Sorry but I don't know what you mean by a demo file. 
The copy of the error report that I've send as details isn't good?
I've tried that, I'm using the latest weekly builds.

maxwxyz, 2024-03-01:
I cannot reproduce this error. If you could provide the file where the error occurs or add steps to reproduce the error, others could try to figure out the error.

cpmach, 2024-03-01:
It happens with any file that i've tried.
Even in ones that I produced in preview versions and used windows and doors. 
It happens when I use the command doors/windows and aply one of those objets to a wall.

luzpaz, 2024-03-01:
Please recreate the issue in freecad, save, zip file and drag and drop into github textbox so we can test the file

cpmach, 2024-03-02:
When I do that freecad crashes and I get an error report.
Is this wath you want?
[freecad_2024-03-02-133913_iMac.crash.zip](https://github.com/FreeCAD/FreeCAD/files/14469537/freecad_2024-03-02-133913_iMac.crash.zip)

cpmach, 2024-03-02:
Or this one?
[Residencial Cavaleiros de Cristo.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14469539/Residencial.Cavaleiros.de.Cristo.FCStd.zip)

Michel2048, 2024-03-03:
Same thing here with Linux Mint AppImage and Flatpack version and Windows 10 64bit installer version (all 0.21.2). App window immediately close when adding windows on wall, no error message.

maxwxyz, 2024-03-03:
I cannot reproduce on Windows weekly build.
<details><summary>Version</summary>
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git)
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

</details> 
I've tried it on a fresh file and it works normally. I've opened @cpmach file but I do not know where you struggle or where the error would occur.
Can you add steps how to reproduce?

Michel2048, 2024-03-03:
I do this simple steps :

1. I create a new document
2. go to sketcher
3. create a rectangle on XY plan
4. go back to arch
5. create a wall from sketch with the wall button
6. select one face
7. click on window button
8. place window with mouse
9. it crash

maxwxyz, 2024-03-03:
No crash at all:
[image]

cpmach, 2024-03-03:
I do exactly the same and it crashes. 
Can it be because you're using 0.22.0 version?
I'm using the last weekly builds for the version 0.21.2.

cpmach, 2024-03-03:
Hi @maxwxyz, the doors and windows that you see in the file I achieve to insert one with no host and the use copies.

cpmach, 2024-03-03:
...and THEN use copies.
sorry

luzpaz, 2024-03-06:
@JohnOCFII do you mind trying to reproduce on macOS (stable and dev) ? TIA

JohnOCFII, 2024-03-06:
> @JohnOCFII do you mind trying to reproduce on macOS (stable and dev) ? TIA

I am not able to reproduce a crash when I follow the steps above in either the release version of 0.21.2 or the current weekly. I also admit that I've rarely launched `Arch` before today, though.

```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3

Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

[image]
[image]

maxwxyz, 2024-03-06:
@cpmach @Michel2048 as you are the only ones being able to reproduce on 0.21, could you please try this in the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

cpmach, 2024-03-06:
I've tried and it's the same.
[freecad_2024-03-06-224021_iMac.crash.zip](https://github.com/FreeCAD/FreeCAD/files/14516602/freecad_2024-03-06-224021_iMac.crash.zip)

JohnOCFII, 2024-03-07:
> I've tried and it's the same.
> [freecad_2024-03-06-224021_iMac.crash.zip](https://github.com/FreeCAD/FreeCAD/files/14516602/freecad_2024-03-06-224021_iMac.crash.zip)

Are you running any 3d party tools on your Mac that might interact with FreeCAD?  We've seen issues in the past with tools like -magnet- or other similar tools that can impact window layouts and/or keyboard entry.

cpmach, 2024-03-07:
No!
The only thing new is the version of freecad.

maxwxyz, 2024-03-07:
@Michel2048 what about you? You seem to be able to reproduce this.

Gitus007, 2024-03-07:
I encountered the same error in 0.21.2-WIN-x64-installer version as well as in 0.22.0dev.36277 (Git) (Windows 10 build 19045, 64-bit, Release). However, with FreeCAD installed on my laptop, the error only appears when inserting the 2nd window into the same wall. Strangely enough, when installing on my desktop, the error usually occurs when inserting the 5th window!

xtemp09, 2024-03-08:
@Michel2048, you wrote that you can reproduce it on Linux Mint. When a crash occurs, Linux prints backtrace into the command-line. Could you upload it here?

xtemp09, 2024-03-09:
@cpmach, @Michel2048, what you are experiencing seems like an OpenGL issue. All OpenGL problems can be easily fixed with installation of _Mesa_. In this case _FreeCAD_ is launched with the following command:

```
[xtemp09@localhost bin] $ LIBGL_ALWAYS_SOFTWARE=true GALLIUM_DRIVER=llvmpipe ./FreeCAD
```

----

Another option that might work sometimes is to disable hardware acceleration. To do that, turn on the checkbox Edit → Preferences → Display → Use software OpenGL and restart _FreeCAD_.

Gitus007, 2024-03-13:
For Windows users it might be a fesable workaround to apply FreeCAD-0.20.2-WIN-x64-portable-2. At least for my model and up to now, I did not get any such errors when using this older portable version of FreeCAD.

Roy-043, 2024-05-15:
This may be related to #13915.

maxwxyz, 2024-05-17:
@cpmach reported the issue is fixed, so I'll close this then.

closed 2024-05-17 by maxwxyz