FreeCAD/FreeCAD #12685: [Problem] v0.22.0 Preferences General Change TreeView Mode - crash MacOS

iAmVaeVictus, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Go to preferences | General 
2. Change TreeView Mode
3. Click Save
4. Crash

Video

https://github.com/FreeCAD/FreeCAD/assets/7082654/f5a48257-d265-4ee5-bc8a-99c969522fec

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AlexSzatmary, 2024-03-03:
I tried to repro on my M2 Mac on 14.0 and it was fine. Can you try again on the latest build, 36135, dated 2024-03-01?

iAmVaeVictus, 2024-03-18:
No, just crashed for me on the following 

OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenDark 2023.12.17

xtemp09, 2024-03-19:
@iAmVaeVictus, what does the backtrace contain?

iAmVaeVictus, 2024-03-19:
> @iAmVaeVictus, what does the backtrace contain?

@xtemp09 - you mean the Apple Error Report, or whatever it is called? It doesn't throw one for some reason for this crash.

iAmVaeVictus, 2024-03-19:
Oh, and, I've upgraded my computer since encountering this error. It is now the following: 

Hardware Overview:

  Model Name:	Mac Studio
  Model Identifier:	Mac14,13
  Model Number:	Z17Z000MELL/A
  Chip:	Apple M2 Max
  Total Number of Cores:	12 (8 performance and 4 efficiency)
  Memory:	64 GB
  System Firmware Version:	10151.81.1
  OS Loader Version:	10151.81.1

luzpaz, 2024-03-20:
@JohnOCFII any chance you have a moment to reproduce this issue?

JohnOCFII, 2024-03-20:
If you don't press `Apply` before `OK` you get the crash.  (As I've mentioned previously elsewhere, the concept of having BOTH an `Apply` and an `OK` button is not very common in macOS.). 

Also it is late, and I'm tired, and I'm pretty sure this is unrelated, but my Mac and FreeCAD are both in dark mode, and FreeCAD is now painting a white screen as it launches after the launch window but before it paints the main welcome / start page.

[image]

```
OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

Crash Report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [14869]
Path:                  /Applications/Other Apps/FreeCAD-36487.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-03-19 22:05:55.3579 -0500
OS Version:            macOS 14.4 (23E214)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       89DBCF32-2178-49C0-A53F-D197D9B6522A

Time Awake Since Boot: 140000 seconds
Time Since Wake:       10458 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000c58
Exception Codes:       0x0000000000000001, 0x0000000000000c58

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [14869]

VM Region Info: 0xc58 is not in any region.  Bytes before following region: 4338938792
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1029f0000-102a10000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-36487.app/Contents/Resources/bin/freecad

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libFreeCADGui.dylib           	       0x10434d344 std::__1::__hash_iterator<std::__1::__hash_node<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, void*>*> std::__1::__hash_table<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::__unordered_map_hasher<App::DocumentObject*, std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::hash<App::DocumentObject*>, std::__1::equal_to<App::DocumentObject*>, true>, std::__1::__unordered_map_equal<App::DocumentObject*, std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::equal_to<App::DocumentObject*>, std::__1::hash<App::DocumentObject*>, true>, std::__1::allocator<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>>>::find<App::DocumentObject*>(App::DocumentObject* const&) + 128
1   libFreeCADGui.dylib           	       0x10462b384 Gui::DocumentObjectItem::~DocumentObjectItem() + 1396
2   libFreeCADGui.dylib           	       0x10462b574 Gui::DocumentObjectItem::~DocumentObjectItem() + 12
3   libQt5Widgets.5.15.13.dylib   	       0x103f7d9fc QTreeWidgetItem::~QTreeWidgetItem() + 300
4   libQt5Widgets.5.15.13.dylib   	       0x103f7afa8 0x103dac000 + 1896360
5   libQt5Core.5.15.13.dylib      	       0x1066f5ad8 QObjectPrivate::deleteChildren() + 108
[... 32 lines cut ...]
38  AppKit                        	       0x197ccc454 __33-[NSApplication runModalSession:]_block_invoke + 112
39  AppKit                        	       0x197abd684 _NSTryRunModal + 100
40  AppKit                        	       0x197ccc324 -[NSApplication runModalSession:] + 148
41  libqcocoa.dylib               	       0x1084483ac 0x10841c000 + 181164
42  libFreeCADGui.dylib           	       0x1044509c0 Gui::Command::_invoke(int, bool) + 516
43  libFreeCADGui.dylib           	       0x104450780 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348
44  libQt5Core.5.15.13.dylib      	       0x1066f9e84 0x1065b4000 + 1334916
45  libQt5Core.5.15.13.dylib      	       0x1066dd094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
46  CoreFoundation                	       0x194035e44 __CFRunLoopDoSource0 + 176
47  CoreFoundation                	       0x194035bb4 __CFRunLoopDoSources0 + 244
48  CoreFoundation                	       0x1940347a0 __CFRunLoopRun + 828
49  CoreFoundation                	       0x194033e0c CFRunLoopRunSpecific + 608
50  HIToolbox                     	       0x19e7cf000 RunCurrentEventLoopInMode + 292
51  HIToolbox                     	       0x19e7cec90 ReceiveNextEventCommon + 220
[... cut ...]
54  AppKit                        	       0x19807edec -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
55  AppKit                        	       0x19787fcb8 -[NSApplication run] + 476
56  libqcocoa.dylib               	       0x1084484d0 0x10841c000 + 181456
57  freecad                       	       0x1029f7030 main + 4932
58  dyld                          	       0x193bce0e0 start + 2360

Thread 1:
0   libsystem_pthread.dylib       	       0x193f51d20 start_wqthread + 0
```

xtemp09, 2024-03-20:
@iAmVaeVictus, if you watch the video you have uploaded, you will see that at _00:08_ MacOS asks you if you liked to report the crash.

Also, in the description you wrote, “_Click Save_”. It somewhat contradicts JohnOCFII's post.

Can you reproduce the crash on the new system?

iAmVaeVictus, 2024-03-20:
@xtemp09 Yes this was reproduced on new system. I do have a crash report today, attached. Must have missed it yesterday. 

[Issue 12685 Apple Crash Report.txt](https://github.com/FreeCAD/FreeCAD/files/14669971/Issue.12685.Apple.Crash.Report.txt)

Steps:
1. Open FC
2. VERY IMPORTANT STEP - Open an existing FCstd drawing.
3. Preferences | General 
4. Change the TreeView Mode (either way, doesn't matter)
5. Select APPLY or SAVE (doesn't matter which)
6. Crash

Works normally if you DON'T open a file first. When you do NOT open a file first, it asks to reopen the app to enable the changes. Seems to only occur after first opening a file. 

Did not happen with an STL file. 

Hope this helps.

AlexSzatmary, 2024-04-02:
I was able to reproduce this with 2 FreeCAD weekly builds including the most recent.

[Issue.12685.Apple.Crash.Report-2.txt](https://github.com/FreeCAD/FreeCAD/files/14830739/Issue.12685.Apple.Crash.Report-2.txt)
[Issue.12685.Apple.Crash.Report-3.txt](https://github.com/FreeCAD/FreeCAD/files/14830738/Issue.12685.Apple.Crash.Report-3.txt)

Step 2 is indeed very important.

If I do not have a part file, the TreeView mode immediately updates in the sidebar upon hitting Apply. After hitting Okay, a dialog box says, "You must restart FreeCAD for changes to take effect."

AlexSzatmary, 2024-04-02:
> If you don't press `Apply` before `OK` you get the crash.

I got the 2 crash reports above having clicked Apply before OK. I also get a crash if I click OK directly.