FreeCAD/FreeCAD #12686: [Problem] [Regression] Inverted Section Marks

frank-trampe, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Importing a file from 0.21.2 into a recent weekly build causes section marks to appear on the opposite of the section line, breaking densely populated views made when they were located on the other side. The old behavior seems slightly better because it tends to place the section identifier away from the center of the drawing and is more likely to place the identifier away from the depicted part. Even if there is a compelling reason to switch, there ought to be an option to control this behavior to avoid breaking existing drawings.

[Past.pdf](https://github.com/FreeCAD/FreeCAD/files/14464543/Past.pdf)
[Present.pdf](https://github.com/FreeCAD/FreeCAD/files/14464544/Present.pdf)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
@WandererFan not sure if this is related to https://github.com/FreeCAD/FreeCAD/pull/12533 and I'm not familiar with ASME. ISO looks like the before drawing.

frank-trampe, 2024-03-01:
As a disclaimer, the cosmetic lines got all jumbled during a load/save cycle, so some of the dimensions are nonsense, but they are still in roughly the right places and show how difficult it can be to reaccommodate the section marks.

WandererFan, 2024-03-06:
Looks from here like you have your preference for Section Line Convention set to ANSI/ASME.   

[image]

frank-trampe, 2024-03-06:
@WandererFan, it was set to ANSI in the weekly build. There is no such option in 0.21.2 as far as I can tell. I changed the setting in the weekly build to ISO, but the arrows and marks are still in the ANSI positions.

frank-trampe, 2024-03-06:
Do I need to save and to reload the file to propagate the change?

WandererFan, 2024-03-07:
The change in preference will not cause an automatic update by itself.  You need to press the recompute button if it is available (not greyed out) or use recompute object from right click in the tree.

luzpaz, 2024-03-09:
What's the verdict here ?

frank-trampe, 2024-03-09:
Refreshing does work. I thought that I tried that before, but I guess not.

frank-trampe, 2024-03-09:
Oh. I know what I did. I tried "recomputing" rather than "redrawing".

closed 2024-03-09 by frank-trampe