FreeCAD/FreeCAD #12687: [Problem] [Regression] Blank display of reopened Techdraw view

frank-trampe, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening a file generated in FreeCAD 0.21.2 in a recent weekly build and attempting to open a previously inactive (invisible, not updated) Techdraw page results in a blank display with just a black dot in the upper left corner. "Fit all" changes nothing. Opening pages that were left active when the file was last saved works. Opening the inactive page in 0.21.2 works, even with "Toggle Updated" left false.

[Base_Framing_2_30_Package_1.zip](https://github.com/FreeCAD/FreeCAD/files/14464625/Base_Framing_2_30_Package_1.zip)

[image]

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

frank-trampe, 2024-03-01:
Setting `KeepUpdated` and `Visibility` to `True` in `Document.xml` and `Visibility` to `True` in `GuiDocument.xml` made FreeCAD load the view correctly. The changes to `Document.xml` were not sufficient and may not have been necessary. I can test this in more granularity if it helps.

WandererFan, 2024-03-07:
If the page's Visibility is false, and KeepUpdated is also false, I would not expect anything to be drawn.   

This is especially true at document load time, as KeepUpdated == false will prevent the building of the drawing geometry. 

I don't think this is a bug.  You've told it not to draw anything, and it isn't.

WandererFan, 2024-05-18:
You don't need to edit the xml file.   Setting KeepUpdated to true and a recompute should update the page. 

ShowDrawing controls whether the Page's tab is visible.  It doesn't affect the content of the Page.

closed 2024-05-18 by WandererFan

frank-trampe, 2024-05-18:
Thanks for looking into this. Per my previous comment, setting "KeepUpdated" to true and recomputing does not update the page, which is why I was editing the file. Do you have any other suggestions?