FreeCAD/FreeCAD #12688: [Problem] V0.22 Sketcher_Dimension: proposed vertical constraint on unconstrained hor line results in error

Roy-043, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a horizontal line.
2. Remove the horizontal constraint.
3. Call Sketcher_Dimension.
4. Select the line.
5. Press M twice.
6. Result: Error message. See below.

```
19:26:20  Updating geometry: Error build geometry(6): Both points are equal
19:26:20  Invalid solution from DogLeg solver.
19:26:20  Updating geometry: Error build geometry(6): Both points are equal
19:26:20  Invalid solution from LevenbergMarquardt solver.
19:26:20  Updating geometry: Error build geometry(6): Both points are equal
19:26:20  Invalid solution from BFGS solver.
```

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git)
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
This has nothing to do with the Dimension tool, it also throws this error when trying to apply the vertical constraint.
Seems like this is because start and end point are on the same V height to be vertical (would be coincident). The same will happen with a vertical line trying to constrain it horizontal.

Roy-043, 2024-03-02:
IMO this has at least something to do with the Sketcher_Dimension tool. It will not iterate over the horizontal and vertical constraint options (when M is pressed) if the edge itself is constrained with a horizontal or vertical constraint, but will offer these options if a line is horizontal because it endpoints are constrained with the same Y coordinate (for example through dimensional constraints or coincident constraints).

FEA-eng, 2024-07-01:
@PaddleStroke

PaddleStroke, 2024-07-03:
There're 2 possible solutions : 
A - Check if line is effectively vertical/hor before offering the constraint. Then skip it if necessary
B - If points are effectively vertical/hor, then still apply the constraint, but move the point slightly so that they are not vertical/hor.

Thoughts?

Roy-043, 2024-07-03:
If you move the points slightly you might get a very short line, which could be problematic. I would go for solution A.

maxwxyz, 2024-07-04:
Every kind could appear short depending on the zoom level and sketch size, so when zooming in, this could look like two distant points

closed 2024-10-28 by yorikvanhavre