FreeCAD/FreeCAD #12689: [Problem] v0.22.0 Tiled Sketch and spreadsheet windows swap location upon click MacOS

iAmVaeVictus, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Tile spreadsheet and drawing view. Drawing on Left. 
2. Edit a sketch in the tree view. Sketch opens on left. 
3. Click spreadsheet. Location of the two windows swaps. This is repeatable. 

Video attached. 

https://github.com/FreeCAD/FreeCAD/assets/7082654/7da0cb8d-691f-41e9-aa9e-eba92323612f

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-01:
I cannot reproduce this on Windows.

luzpaz, 2025-06-14:
@johnocfii any chance you can test on macos ?

JohnOCFII, 2025-06-14:
> [@JohnOCFII](https://github.com/JohnOCFII) any chance you can test on macos ?

I'm unable to duplicate, but before trying this test, I'd never used tiled windows in FreeCAD before, so maybe not the best test.  The video was pretty straightforward, so I think I did it correctly.

I started up a separate user account so I could use the default Classic View without overlay to do the test.

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42149 (Git)
Build date: 2025/06/13 08:44:01
Build type: Unknown
Branch: main
Hash: 9714e2f87a6240d1b2b979eb39c7d34791aafdae
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.6.1
  * Telemetry 1.0.1
```

closed 2025-08-29 by github-actions[bot]