FreeCAD/FreeCAD #12690: [Problem] [Regression] Item attributes in feet when set to inches

frank-trampe, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If one sets the "Default unit system" to "US Customary (in, lb)" or "Imperial Decimal (in, lb)", values over 12 inches show in feet. This makes design of small parts and laying out objects on an Arch C sheet difficult. The settings worked correctly in 0.21.2.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Core

### Anything else?

![Screenshot from 2024-03-01 15-31-09](https://github.com/FreeCAD/FreeCAD/assets/2512464/8c313958-e4c0-4ab2-9986-ad8922a65812)
![Screenshot from 2024-03-01 15-32-12](https://github.com/FreeCAD/FreeCAD/assets/2512464/101af008-cdba-45be-8b76-a15b53952e80)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

frank-trampe, 2024-03-01:
This is a bit more complicated than I thought. The project-level units (which I was not overriding) somehow got changed when I was migrating the file. When I override project-level units, it works properly. I cannot find any documentation on viewing or setting project-level units. How does one do that?

WandererFan, 2024-10-19:
[image]

WandererFan, 2024-12-20:
Imperial decimal does not use feet for larger numbers. 

I would say that this is a documentation problem (re how to set project level info), not a bug.

closed 2025-06-29 by github-actions[bot]