FreeCAD/FreeCAD #12691: [Problem] Inverse Shortening of Cosmetic Lines

frank-trampe, 2024-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempting to shorten a cosmetic line in TechDraw makes the line longer. This problem is present in 0.21.2 also.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-02:
Can confirm on Windows too

frank-trampe, 2024-03-05:
Huh. That's odd. Shortening those works.

WandererFan, 2024-03-06:
Is this using one of the extension functions (red icons)?

frank-trampe, 2024-03-06:
@WandererFan, yes. The one with the drop down that allows selecting shortening or lengthening.

WandererFan, 2024-06-10:
fixed as part of #14216.

closed 2024-06-10 by WandererFan