FreeCAD/FreeCAD #12692: [Problem] The Supressed property field has no Fx icon

macdroid53, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Pad/Pocket/etc. have a Boolean property "Suppressed". The property field has no way to enter an expression.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git) AppImage
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.7
  * Defeaturing 1.2.2
  * Help 1.0.3
  * kicadStepUpMod 10.23.2
  * Assembly3 0.12.2
  * Curves 0.6.23
  * dodo 1.0.0
  * DynamicData 2.60.0
  * freecad.gears 1.2.0
  * fasteners 0.5.12
  * A2plus 0.4.63
  * Assembly4 0.50.7
  * sheetmetal 0.4.2
  * CurvedShapes 1.0.5
  * OpenDark 2023.12.17
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-02:
screenshot would be nice. please add to the original post. tia

macdroid53, 2024-03-02:
[image]

Syres916, 2024-03-02:
I'd flip the question and ask should all _App::PropertyBool_ have the ability to use expressions as the Suppressed property is no different from say ModelThread in PD>Hole or any other boolean?

macdroid53, 2024-03-02:
I don't know about other bools, but, being able to set, for an example, an expression in this property that is a DynData or spreadsheet cell would be quite handy.

maxwxyz, 2024-03-02:
@FlachyJoe could you weigh in please

FlachyJoe, 2024-03-02:
@maxwxyz please assign me

FlachyJoe, 2024-03-02:
@macdroid53 As I know, we currently can't add the _fx_ icon in the true/false combo box. But you can access the property expression with a right-click.
[image]

You or @maxwxyz can close this issue as not planned. A more general issue about boolean properties can be open if you think it's needed.

closed 2024-03-02 by maxwxyz

macdroid53, 2024-03-02:
> @macdroid53 As I know, we currently can't add the _fx_ icon in the true/false combo box. But you can access the property expression with a right-click. [...]

Totally forgot you could do that.