FreeCAD/FreeCAD #12700: [Problem] Recompute enable/disable setting should be saved in the FCStd file

maxwxyz, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session.
**Recompute enable disable setting should persist over a file save**

Remarks:
When you have a large model that uses spreadsheet, you have to disable the recompute to make it fast. When you close the file and open it again, the recompute is enabled again. So you have to turn it off again.
The enable and disable recompute setting does not persist when you save a file, it's not saved with the file.

Related issues to recompute:

- https://github.com/FreeCAD/FreeCAD/issues/6250
- https://github.com/FreeCAD/FreeCAD/issues/6233
- https://github.com/FreeCAD/FreeCAD/issues/11699

### Full version info

```shell
0.22
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Trallala-Hobsassa, 2024-10-03:
Sorry for issue #16911. I searched for this request for over half an hour, but found nothing. (But I'm also new to GitHub...) So sorry again.

If you ever tackle this feature, please don't forget to make the follow-up function "Allow partial recomputes" storable, too.

Thank you and best regards
Trallala-Hobsassa!

maxwxyz, 2024-10-03:
just add it as comment to the other issue so it won't get lost