FreeCAD/FreeCAD #12701: [Problem] Core: Tree reordering not consistently possible

maxwxyz, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/pull/12293 introduced a solution to https://github.com/FreeCAD/FreeCAD/issues/5907 and while trying if this could also solve #12378 I had weird behaviors when trying to reorder tree items.

- reordering only works sometimes
- reordering sometimes toggles visibility of the object (https://github.com/FreeCAD/FreeCAD/pull/12298)
- reordering does not work in all WBs (e.g. not in TechDraw
- trying to drag between tree items does only work in some cases (no visual indication on most attempts). Items can accidentally be dragged into other items where it does not make sense (a Part object into a Arch Wall) as it not possible to reorder them. Once inside, there is no possibility to drag it out of it.
- a Part Design feature of a body can be dragged out of it and break the model with no option to drag it inside again (e.g. drag out a chamfer)

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/56b0dce0-9086-4203-8c77-e07f27eb04ec

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git)
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/FreeCAD/FreeCAD/issues/11676

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-02:
@PaddleStroke FYI

PaddleStroke, 2024-03-13:
Ah many issues in one here.
So https://github.com/FreeCAD/FreeCAD/pull/12293 only did the following : 
- enable reorder at root level
- enable reorder in App::Parts and App::DocumentObjectGroups
If an object wants to enable re-order (techdraw for instance) then its view provider needs to reimplement the function 

`    bool acceptReorderingObjects() const override { return true; };`

kinghat, 2024-03-13:
i think when dragging around at the top level the line that shows where its going to be placed could be improved. most of the time i dont see it at all. this makes it hard to drop things around objects that can be dropped into. a body cant be dropped into a body but a body can easily be dropped into a part container for example. especially when trying to drag to the bottom of the tree or to the bottom of an expanded body tree.

App::DocumentObjectGroups include bodies?

PaddleStroke, 2024-03-13:
Yes the line does not appear when reordering on root level. This is very annoying bug I have been chasing (see discussion in the PR).

nazar-pc, 2024-07-07:
I'd like to be able to reorder things inside of `Part`. I was able to drag sketch into it, but now it is at the end of the list of elements and there is no way to move it into more logical place without redoing elements of the part.

luzpaz, 2024-10-26:
@maxwxyz should we split this ticket up ?

maxwxyz, 2024-10-26:
@luzpaz I'm closing this, it is too vague and not creating any bugs.

closed 2024-10-26 by maxwxyz

luzpaz, 2024-10-26:
@maxwxyz Are you sure? It's not super vague like:
* reordering does not work in all WBs (e.g. not in TechDraw)

Is that still true?

maxwxyz, 2024-10-26:
Reordering TD pages works now