FreeCAD/FreeCAD #12705: [Problem] File templates are only available on the start page

pierreporte, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The start page has buttons to create files with a template: Standard Part, Assembly, 2D Draft, and BIM/Architecture. When it is closed, these templates are not available anymore. They should be present in the File menu as well for easy access. Here is a proposition:

```
File ↓
    New → Empty file (Ctrl+N)
          -------------------
          Standard part
          Assembly
          2D Draft
          BIM/Architecture
          -------------------
          Template...
    Open... (Ctrl+O)
    Open Recent → ...
    -----------------
    ...
```

I added a “Template…” to give an idea of how it would look if there was a template system, as briefly discussed in #12565. The last line would open a window with the full list of templates while the normal lines would be the pinned templates.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git) AppImage
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * TabBar (Disabled)
  * Curves 0.6.23
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.0
  * OpenDark 2023.12.1
  * OpenLight 2023.12.17
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
To add: Maybe the corresponding WB could be activated when creating a new file from a template.
Activate WB Assembly for new assembly, Part Design for new part, BIM for BIM,...

pierreporte, 2024-04-22:
@maxwxyz for the DWG project. Would be a good thing to do since the start page won’t be the default anymore apparently.