FreeCAD/FreeCAD #12707: [Problem] v0.22.0 Typing dimension values changes view instead MacOS

iAmVaeVictus, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Typing in values for new shapes using the new functionality instead changes the view. 
1. Start a sketch
2. Select a shape (e.g. box), click first starting point. 
3. Try to type a value in the dimensional box that is already highlighted. 
4. Instead of entering the value, the view changes. 

Must click the second point to finish the box and then go back and change the values of the shape. Was working in earlier version of 0.22.0 but with this version 36117 downloaded on Monday ( Feb 26, 24), it is broken. 

May be a MacOs only issue. Seems like you'd get notice for this one pretty quickly but I don't see one. 

### Full version info

```shell
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36117 (Git)
Build type: Release
Branch: main
Hash: 8a3407aee6dff8218282d9cdc3f4e32e7ad958d4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
I cannot reproduce this.
<details><summary>Version</summary>
<p>

```
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35153 (Git)
Build type: Release
Branch: main
Hash: ade8664717957cc5b02cb13f6f3043e4be3b341f
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.2.0
  * OpenDark 2023.12.17
  * CfdOF 1.25.2
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.12
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.2
  * 3D_Printing_Tools
  * Curves 0.6.23
```

</p>
</details>

JohnOCFII, 2024-03-04:
I can not reproduce this with the current weekly.  

One item to be aware of is that, like in many places in FreeCAD, the mouse (or in my case, the TouchPad on my MacBook Pro) remains active, and impacts the data entered in the fields, so an errant touch will cause very strange behavior.  As mentioned elsewhere, I (and others) have recommended that mouse movement (or Trackpad input) should somehow be disabled when a data field is set to receive input.

John

```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36135 (Git)
Build type: Release
Branch: main
Hash: e99220daf62069be653f73e720bcad9e0cd2fb97
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

luzpaz, 2024-03-06:
My bad. I misread John's reply.

> One item to be aware of is that, like in many places in FreeCAD, the mouse (or in my case, the TouchPad on my MacBook Pro) remains active, and impacts the data entered in the fields, so an errant [...]

@JohnOCFII should we open a new ticket then that is macOS specific ?

JohnOCFII, 2024-03-06:
> @JohnOCFII should we open a new ticket then that is macOS specific ?

I'm not sure that it **IS** macOS specific (I guess I can test in Debian Linux to confirm) but it is most likely experienced by those with laptops and trackpads where it is not uncommon to rub against the trackpad when reaching for the number keys to enter data.  In many ways I wonder if the issue (which impacts MANY areas of input in FreeCAD) is similar to this?  I had added a comment in this Issue specific to Path/CAM workbench, but this issue is apparent in many areas where input fields are active. I'll test in Debian on this laptop to see if behavior differs in that environment.

https://github.com/FreeCAD/FreeCAD/issues/11807#issuecomment-1974747504

JohnOCFII, 2024-03-06:
> @JohnOCFII should we open a new ticket then that is macOS specific ?

I did just re-test this specific issue in both MacOS and Debian Linux and behavior is the same in both. I think this issue can probably be closed.  

The data input fields DO accept mouse movement (or trackpad movement) as "ENTER" and thus ends data input.  Not sure if that is desired (as opposed to requiring a keyboard press of ENTER or RETURN to accept the end of entry.

As to the other issue -- if I create something separate, I'll probably refer to the item I mentioned above (#11807 )

iAmVaeVictus, 2024-03-06:
Testing on the build from 03/04 (36227), I agree that this is FIXED. It does not behave like it was where it was not accepting my input from the keyboard (and no, it was not errant touching of the mouse). It was simply broken for a few builds.

closed 2024-03-06 by maxwxyz