FreeCAD/FreeCAD #12708: [Problem] Assembly: Parts created in context can’t be saved in a separate file

pierreporte, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When parts are created with `Std_Part` directly in an existing Assembly container, they are stored in the document containing the assembly (the document can be on the disk or in memory before being saved). It is not possible to tell FreeCAD that it needs to be in a separate document. To store parts in independent documents, they need to be created separately and not in context.

This limits the ability to work with several files. While it is understandable to allow for the whole project to be in a single file, this is not the most common in the industry. Even then, it may be desired to extract one part of a project to reuse it elsewhere more easily.

When the part is detached, it must be transformed into a link in the assembly and all other links to existing geometry must be replaced as well so the process is as seamless as possible.

There should be a program option to automatically detach parts (and later sub-assemblies) from their parent, for those who prefer to work with one component per file.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git) AppImage
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * TabBar (Disabled)
  * Curves 0.6.23
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.0
  * OpenDark 2023.12.1
  * OpenLight 2023.12.17
```

### Subproject(s) affected?

Assembly

### Anything else?

See discussion #11742

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luisDias89, 2024-03-03:
Stated clearly and concisely, that's exactly what FreeCAD lacks. Congratulations!

PaddleStroke, 2024-03-13:
So this issue is also rather a App::Part issue. It's all revolving around the idea of single file per document it seems.

PaddleStroke, 2024-03-13:
If we use Std_Part, then we are modifying the behavior for all parts.
If we make another tool in assembly, then we have a duplicate tool on the UI. Which is going to be very confusing.

Maybe the Std_Part should prompt the user 'Create part in a new document?' yes no always yes always no.

pierreporte, 2024-03-13:
Modifying `Std::Part` could be a good thing because assemblies inherit from it so it would allow for sub-assemblies to be saved in a separate file as well, which is needed. A separate tool may be interesting, considering that `Std::Part` is actually `Std::Component` from which parts proper and assemblies are derived. The `Std::Part` (renamed or not) would disappear from the toolbars.

I like your dialog but I would transform the always buttons by a single checkbox “never ask again”. The label could say that it can be changed in the settings.