FreeCAD/FreeCAD #12709: [Problem] Assembly: `Std_Part` doesnt create the part as a child of the active Assembly container

pierreporte, 2024-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:

1. Create an assembly with `Assembly_CreateAssembly`.
2. Activate the assembly if it is not already active.
3. Create a part with `Std_Part`.
4. The part is at the same level of the assembly.

Expected result: the part is a child of the assembly.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git) AppImage
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * TabBar (Disabled)
  * Curves 0.6.23
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.0
  * OpenDark 2023.12.1
  * OpenLight 2023.12.17
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luisDias89, 2024-03-03:
Exactly, and be able to make "joins" between parts and assemblies

PaddleStroke, 2024-03-13:
This is actually a App::Part issue. If you have a App::Part that is active and click on Std_Part, the new part is on the root level.

Assembly object derives from App::Part.

So the question is, do we want to modify the behavior of Std_part so that it creates a part in the active part if any is active?

pierreporte, 2024-03-13:
@PaddleStroke Creating parts in context in an existing assembly is a very common operation. The ability to do so is a must have. Note that the origin of the new part should be coincident with the assembly origin by default.

PaddleStroke, 2024-03-13:
I don't say it shouldn't be done. Just wondering what's the way to do it.
If we use Std_Part, then we are modifying the behavior for all parts.
If we make another tool in assembly, then we have a duplicate tool on the UI. Which is going to be very confusing.

PaddleStroke, 2024-03-13:
Maybe the Std_Part should prompt the user 'Create part in a new document?' yes no always yes always no.

PaddleStroke, 2024-03-13:
Sorry I mistook this with #12708
For this issue I don't say we should not do the change, just that this will change behavior for all App::Parts. Which maybe is fine.

pierreporte, 2024-03-13:
Depending on how sub-assemblies work in your follow-up PR, it may be a good idea to change `Std::Part` for the same reasons described in #12708.

PaddleStroke, 2025-11-27:
Actually I think that this issue and #12708 should both be closed since now we have the `Assembly_InsertNewPart` command. As this command does exactly what you want : 
It creates a App::Part in a separate files and adds a link in the assembly

pierreporte, 2025-11-27:
We can argue about this issue, indeed there is now a command to create a new Part container in the active assembly. However having two commands to create this container is bad UX and I still think that there should be only one that can do both.

I absolutely disagree about #12708 being contradictory with this issue, because #12708 is about detaching an existing Part container that has been created in a file (typical use case is within an Assembly container) into a new separate file where it would be the root, and creating a link to this new file in the original one.