FreeCAD/FreeCAD #12710: [Problem] - App launch fails - stack smashing detected

hakiya, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

```
hakiyama513@penguin:~$ freecad
FreeCAD 0.20.2, Libs: 0.20.2R
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

*** stack smashing detected ***: terminated
Aborted (core dumped)
hakiyama513@penguin:~$ 
```

### Full version info

```shell
FreeCAD 0.20.2, Libs: 0.20.2R

ARM64 ChromeOS
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hakiya, 2024-03-06:
Unfortunately, because I am unable to open FreeCAD, I am not able to retrieve the about info. Not sure if there is a way I can get the same about info without loading FreeCAD.

I tried extracting the appimage on my device, and I still see the following:

```
hakiyama513@penguin:~/misc$ ./FreeCAD_weekly-builds-36277-2024-03-04-conda-Linux-aarch64-py310.AppImage --appimage-extract
hakiyama513@penguin:~/misc$ cd squashfs-root
hakiyama513@penguin:~/misc/squashfs-root$ ./AppRun 
FreeCAD 0.22.0, Libs: 0.22.0devR36277 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

*** stack smashing detected ***: terminated
./AppRun: line 43:  2527 Aborted                 (core dumped) "${MAIN}" "$@"
```

Thanks

luzpaz, 2024-03-08:
Can you reproduce with flatpak ?

hakiya, 2024-03-09:
Tried the following, not able to install flatpak with flatpakref:

```
hakiyama513@penguin:~/misc$ flatpak install --from org.freecadweb.FreeCAD.flatpakref --user
error: No such ref 'app/org.freecadweb.FreeCAD/aarch64/stable' in remote flathub
```

luzpaz, 2024-03-09:
Darn, that's right. FreeCAD flatpak doesn't support arch64 yet per https://github.com/flathub/org.freecadweb.FreeCAD/issues/46

luzpaz, 2024-08-17:
@maxwxyz any idea what to do here ?

maxwxyz, 2024-10-07:
@luzpaz  @hakiya https://github.com/flathub/org.freecadweb.FreeCAD/issues/46 got merged now so I guess this is solved. Closing this but please check and reopen if still an issue.

closed 2024-10-07 by maxwxyz