FreeCAD/FreeCAD #12713: Receive error creating additive primitive on plane in 0.21.2, works fine in 0.20.1.

Druloric, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

<class 'SyntaxError'>: ('unterminated string literal (detected at line 3)', ('<string>', 3, 59, "App.getDocument('Unnamed').getObject('Box').Height='2.38 ''", 3, 59))

Creating a padded base (.75" x 9.25" x 28") after which adding an additive primitive (.75" x 9.25" x 28.5") to the surface receives this error.  The same attempt works fine in 0.20.1 release.

### Full version info

```shell
[code]
OS: Linux 6.7.7-arch1-1 (qtile)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
Could you try this in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Roy-043, 2024-03-03:
If the unit system is Building US this may be related to #11345.

maxwxyz, 2024-06-24:
@Druloric could you provide feedback?

MiqCG, 2024-08-26:
Hi,

I have tested with FreeCAD 0.22 revision 38459: the pop-up window with the error message doesn't appear and the Additive Primitive Box can be created on the face of the padded sketch.
Due to lack of information I guessed that was a Primitive Box with US Customary units and no particular orientation of the parts, constraints and base face.

I attach the files created with 0.20.2, 0.21.1 (unsuccesful) and 0.22 rev 38459.
[Primitive.zip](https://github.com/user-attachments/files/16752061/Primitive.zip)

Image with the successful operation in 0.22
[image]

Image of the error with FreeCAD 0.21.1:
[image]

maxwxyz, 2024-08-26:
Thanks I'll close it then as fixed in 0.22

closed 2024-08-26 by maxwxyz