FreeCAD/FreeCAD #12717: Imported image do not respect the `OnTopWhenSelected` attribute

rostskadat, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When 2 images (JPG) are imported, and the `OnTopWhenSelected` has the `Disabled` value set for both image, I would expect part of one of the image that is underneath to not be visible in the orthographic view. However, the image is entirely visible, thus hiding part of the second image that would be otherwise visible. I attach a sample file to demonstrate and the corresponding screen capture. This does not seem to be happening in 0.20 (I'll try to check it to make sure).

### Full version info

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD-geodata 0.1.0
  * FreeCAD-SH3D 0.0.1
  * Render 2024.1.23
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rostskadat, 2024-03-03:
[OnTopWhenSelected.zip](https://github.com/FreeCAD/FreeCAD/files/14472252/OnTopWhenSelected.zip)
[image]

maxwxyz, 2024-03-03:
Could you try this in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

wwmayer, 2024-03-03:
For me it works as expected (tested with v0.22 and v0.21). Maybe it's a graphic driver issue on your machine.

rostskadat, 2024-03-04:
Hello @wwmayer , @maxwxyz 

Thanks for your quick reply. I tried on the latest weekly build:

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git) AppImage
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

and it seems to work fine. I'll proceed by closing the ticket

closed 2024-03-04 by rostskadat