FreeCAD/FreeCAD #12719: [Problem] TechDraw: Surface Finish Symbol Issues

maxwxyz, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Improvements for the surface finish symbol in TechDraw to support older drawings acc. to **ISO 1302** and the new standard **ISO 21920** (which replaced ISO 4287, ISO 4288, ISO 13565-2, ISO 1302)
- It seems there is no way of adjusting/correcting an existing surface finish symbol in TechDraw. The workaround is to delete it and place a new symbol.
- The surface finish symbol (using ISO mode) always have to have a "Ra..." entry under the line. There is no was to remove this label
- The only way to specify the roughness in ISO mode is using the predefined "Ra..." entries from the drop down menu. There is no way to enter a custom dimension, e.g. *Rz 1-3*. Also, only `Ra` is available, no `Rz`,`Rt`, `Rp`, `R_max` or `R_mr` values can be added
- Option `P` for no groves is missing in the dropdown on the bottom
- It seems there are spaces for the predefined dropdown values of Ra: (https://forum.freecad.org/viewtopic.php?t=85758)
- The specified default font in the preferences is not used for this symbol
- The surface finish dialog could see some improvement, not sure it this is only due to @obelisk79 OpenDark theme: https://github.com/obelisk79/OpenDark/issues/34
- New surface finish symbols are not supported (e.g. line above triangle missing) [image] [image]
- New specification of ISO 21920 is not supported [image] [image] [image] https://guide.digitalsurf.com/en/guide-iso-21920-parameters.html

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
@WandererFan FYI, if you need info from the standards we can chat.

WandererFan, 2024-03-06:
I thought there was discussion at the time this tool was implemented that most of the options were not actually used in practice.   @edi271 is the expert on this one.

edi271, 2024-03-07:
I communicate with @maxwxyz. We will try together to improve the tool.

FEA-eng, 2024-04-03:
I would also add that the text is not rotated with the symbol: https://forum.freecad.org/viewtopic.php?t=86667

From SolidWorks:

[image]

sebaras, 2024-05-22:
There is another Failure in Surface Symboles. It must have the possiblity to leave inputsfield Ra emty, when ISO is selected.
Surface symbol for “no chip removal”.
[image]

benj5378, 2024-08-28:
Requirements to do surface finish symbols: keyboard, mouse and **microscope**.

Would anyone mind if I changed the buttons to enlarge the icons? @WandererFan 

[image]

FEA-eng, 2024-09-22:
Drop-down lists overlap here:

[image]