FreeCAD/FreeCAD #12720: [Problem] UI themes in Linux do not look good

Beep6581, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey

Thank you for your hard work on FreeCAD, it's great having a libre CAD option.

I wanted to bring to your attention a theme issue. I experience it using KDE in Manjaro, though it's probably the same in any distro or window manager.

Though stylistically dated, FreeCAD 0.21.2's UI is functional. However, the UI themes in the 0.22.0dev AppImage builds all seem to have functional deficiencies which impede usage of the software.

Notice the issues depicted in the following screenshots:
- Lack of contrast between dark items, including the actual objects in the 3D view.
- Grayed-out icons are difficult to see. Fix: Though grayed-out, they should be clearly legible.
- Undo/Redo are for some reason a different faded color. Fix: If grayed-out, they should have the same color as other grayed-out buttons. If clickable, they should have the same vibrant colors as other clickable buttons.
- Tooltips have no outline and same background color, so they blend into the background. Fix: Tooltips should stand out as such by having an outline and different tooltip background color.
- Frames have no outlines. Fix: Frames should have outlines or be designed in such a way that their background and padding colors make it clear where a frame begins and ends.
- Frame buttons (toggle overlay/toggle floating window/Close dock window) are difficult to see. Fix: Same as for other buttons.
- Excessive padding between icons. I may have a large screen, but I don't want to waste it on padding.

This is the Classic theme:
[image]

Dark:
[image]

Light:
[image]

OpenDark:
[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git) AppImage
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * fasteners 0.5.11
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.5
  * Curves 0.6.23
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
@MisterMakerNL @obelisk79 FYI

Beep6581, 2024-03-03:
Too much icon padding.

Classic:
[image]

ProDark:
[image]

Dark (good padding):
[image]

obelisk79, 2024-03-04:
We are treading into opinion territory, however I'll say this. The extremely low, nearly non-existent padding in Dark where you claim is good padding, is in my  opinion quite wrong. It is difficult to parse so many icons on screen at once, and by having too little spacing between them, the average user will struggle to scan and find what they need because the icons don't have enough white space around them to allow them to serve as a visual target/anchor. 

Clearly, as with yourself, this isn't a problem for some people, but that isn't representative of the average. For the toolbar icons for save, undo/redo etc. There's some odd issue where these icons are pulled from the parent system. I have a similar problem on my KDE linux system as well, not easy to resolve directly using a theme, but I'll gladly look into it when I get back into bug triage for opendark.

luzpaz, 2024-03-08:
JFYI, @Beep6581 is one of the core devs of RawTherapee ;)

MisterMakerNL, 2025-08-29:
> Updated the ticket to contain checkboxes and ability to reference any known open issues. 
> 
> cc @MisterMakerNL what do you think, should this ticket be split into separate issues ?
There is maybe 10% left of the stylesheet from a year ago.😅

luzpaz, 2026-09-22:
@Beep6581 can you review this against the recently merged https://github.com/FreeCAD/FreeCAD/pull/32827 (should be available in the upcoming dev build (**Not** weekly-2026.09.16))