FreeCAD/FreeCAD #12721: [Problem] Selecting in the 3d view when using ellipsis dialog in Propeties view

macdroid53, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Reference this forum thread: https://forum.freecad.org/viewtopic.php?t=85800

When changing properties like Profile, etc. where the property value is a face or edge, you can select the ellipsis in the field an you get a dialog where an entity can be added or changed. You can clear, reset etc. Sometimes you can select in the 3D view a new entity, sometimes not. In the later case you can, if you know the name of the new, for example, face, you can type it in. Also, the dialog is named Link, this seems wrong.

When typing in the field, sometimes the field has to loose and gain focus for the change to have effect.

### Full version info

```shell
This has been a long time behavior.

But noted recently here:
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36230 (Git)
Build type: Release
Branch: main
Hash: 31b12ae9e2fa0cd660e65b69a93c933e2791968c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * OpenDark 2023.12.17
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Curves 0.6.23
  * ArchTextures
  * Nodes 0.1.36
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.2
  * Silk 0.1.5
  * Freecad-Built-in-themes-beta 1.2.2
  * kicadStepUpMod 10.23.2
  * DynamicData 2.60.0
  * workfeature-macro
  * fasteners 0.5.12
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Anything else?

See forum thread for a sample of one dialog of this nature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chrisb-github, 2024-03-03:
Selecting something in 3D view places the - usually much bigger - main window before the dialog, so that the dialog simply seems to be lost. It doesn't come back to front, even not if it is invoked a second time.

Shkolik, 2024-10-03:
It would be very beneficial to let user select sub element in a 3d space, and not just type it's name in a field. It's so inconvenient right now... especially when TNP not polished yet I need to change references very often.