FreeCAD/FreeCAD #12723: [Seatching model tree does not work for numbers]

PCS-Rimage, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I search for alphabetic string in an object then search works, if I search for numeric string then search does not show any results.

Thanks
Phil

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.62
  * Assembly3 0.12.2
  * CurvedShapes 1.0.5
  * freecad.gears 1.2.0
  * Help 1.0.3
  * lattice2 1.0.0
  * pcb 6.2023.1
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.1
  * Silk 0.1.5
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-03:
I can reproduce this on latest dev 
<details><summary>Details</summary>
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git)
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```
</details> 
Right click in the tree and search, it won't show any results for numbers.

maxwxyz, 2024-04-01:
Thanks, go ahead!

FarleyRunkel, 2024-04-10:
Fortunately, I later figured it out and pushed a commit as above. 
: )

closed 2024-04-22 by chennes