FreeCAD/FreeCAD #12724: [Problem] Addon Manager fails to load macros

johnsonm, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Tools → Addon manager

Select a macro, get error `Failed to download data from  -- received response code 0.` and a backtrace in the console:

```
15:39:30  Traceback (most recent call last):
15:39:30    File "/tmp/.mount_freecaauSZi8/usr/Mod/AddonManager/AddonManager.py", line 788, in table_row_activated
15:39:30      self.package_details_controller.show_repo(selected_repo)
15:39:30    File "/tmp/.mount_freecaauSZi8/usr/Mod/AddonManager/addonmanager_package_details_controller.py", line 117, in show_repo
15:39:30      self.update_macro_info(repo)
15:39:30    File "/tmp/.mount_freecaauSZi8/usr/Mod/AddonManager/addonmanager_package_details_controller.py", line 196, in update_macro_info
15:39:30      self.worker.readme_updated.connect(self.macro_readme_updated)
15:39:30  AttributeError: 'PackageDetailsController' object has no attribute 'macro_readme_updated'
```

### Full version info

```shell
OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git) AppImage
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
@johnsonm I've noticed I cannot click on macros which have (wiki) after their name. Is this related or are you not able to click on any macro?

johnsonm, 2024-03-09:
Inconsistent. None of the git-source macros I tried had the "Failed to download data from  -- received response code 0." But they don't all act the same:
* Aerofoil works, showing content and not throwing an error
* Camera shows content **and** throws the same `object has no attribute 'macro_readme_updated'` backtrace but without the "Failed to download data" error.
* BoxCreator does **not** show content and does not throw the error
* DeepCopy works, showing content and not throwing an error

All the wiki-sourced macros fail, throwing the "Failed to download data" error.

So I assume there's actually more than one problem here.

PR #12858 (2024-03-11): Addon Manager: Fix README fetch for macros
When no macro metadata is downloaded, the macro has no URL yet. Fixes #12724