FreeCAD/FreeCAD #12742: [Problem] Title bar is wasted screenspace.

MisterMakerNL, 2024-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most modern programs, especially in Windows uses the title bar for tabs or for the default menubar. In Freecad this is currently not possible.
It looks like this is doable by using dependency called Qgoodwindow.
https://github.com/MisterMakerNL/QGoodWindowFreecad
It works in Windows,Linux and Mac so it sounds like ideal option. 

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-03-04:
What you are asking for is client-side decorations: Basically the title bar is not drawn anymore, and the app provides its own "top" widget that behaves like a title bar (it moves the window when drag/dropping, and it has min/max/close buttons. This is already supported natively by Qt: https://www.qt.io/blog/custom-window-decorations

I also think we should allow that (optional of course, some people will still want the standard title bar, for XX reasons)

I asked about that feature at the FreeCAD day, and everybody there found that an excellent idea. So I think we should do it :) What should still be done is to decide what goes on the title bar. At the same FreeCAD day, the suggestions were:

* The workbench selector
* The MDI tabs
* A "hamburger menu" with the contents of the menu bar (the menu bar is then not displayed when in client-side mode)
* Minimize & close buttons

xtemp09, 2024-03-04:
@PaddleStroke, have you tested _QGoodWindow_ on both _Wayland_ and _X_?

yorikvanhavre, 2024-03-04:
> Now how to achieve this? I would suggest that in the custom 'title bar', we put 'docking areas', one on the left and one on the right. And these custom 'docking area' would let the user drag and drop toolsbars in and out of there. Giving the user full customability.

That's interesting...

I still fear there might be a lot of stuff on the titlebar then :)

Here is my idea:

[image]

The search bar is just taken from thunderbird... But that could be a customizable area indeed where workbenches could define stuff...

PaddleStroke, 2024-03-04:
@xtemp09 I have not done any test on this subject so far. @yorikvanhavre is working on it as far as I know.

@yorikvanhavre interesting idea. The search box is nice.

realthunder, 2024-03-04:
The toolbar drag and drop on to status bar is implemented in this [commit](https://github.com/realthunder/FreeCAD/commit/d15bc174739). This relies on `QStatusBar.addWidget()`. There is no similar API for title bar. In fact, title bars are supplied by native UI platform. You'll need write a completely new widget to replace the title bar.

maxwxyz, 2024-03-10:
I also like @yorikvanhavre idea. The search bar would be nice (similar to https://github.com/FreeCAD/FreeCAD/issues/8290). Also the tabs on top like in most applications. Not sure if this has to be another optional setting in the preferences.

pierreporte, 2024-03-10:
I like what Solidworks do with the title bar. The menu (file, edit, etc.) lies in it and is followed by commonly used icons (new file, open file, save, undo, redo, recompute, settings, and a few others). A search feature is at the right end of the bar. The menu is foldable into the software name. At the center, there is the current file name:

[image]

[image]

I prefer having tabs at the bottom of the screen so that they are out of the way, not mixed with the features while still readily available when needed.

In the Discord server, we briefly discussed about a workspace feature similar to what Fusion 360 provides, to filter workbenches (an Architecture workspace with Arch/BIM/Draft/…, a Mechanical Design with Assembly/PartDesign/Part/…, a Technical Drawing workspace, a Simulation workspace, etc.). The selector could be in the title bar. Though this workspace thing only works with a ribbon interface (be it a full-featured one or lightweight like what @PaddleStroke proposes).

yorikvanhavre, 2024-03-11:
> Not sure if this has to be another optional setting in the preferences.

We cannot just "add stuff to the title bar". Because the title bar is not controlled by FreeCAD, it's controlled by the OS. So the solution here to implement this, is to tell the OS "Do not draw a title bar" and add a widget ourselves that looks like and behaves like a title bar. Some users won't want that, because they might loose some functionality offered by the OS (some linux desktops do pretty interesting stuff with the title bar) and it might also look different than other apps (the title bar will not be themed by the OS anymore). So IMHO this needs to be user-defined and revertable.

PaddleStroke, 2024-03-11:
@yorikvanhavre are you working on this feature yet?

AthidhiVijai, 2024-03-12:
Hi ,
thank you @PaddleStroke @realthunder @yorikvanhavre and all contributors for the efforts you are putting to make UI very beautiful and easy to learn and get job done like me students,
my humble request to all of you that can you make tool bar / workbench tool bar area to toggle overlay and toggle transparent, This functionality is similar to the existing transparency toggle for the Combo View and Property View panels.
it gives at least 12.5 mm to 25 mm area to see the model bit more with clarity and full model in smaller screens like me 13inch laptop 
Thanks in advance and for your valuable time to help the needy 
sincerely
Vijai

PaddleStroke, 2024-03-13:
@yorikvanhavre I am not sure how to do this actually. Should we put QGoodWindow as a 3rd party module? Or we just copy paste what we need?
Perhaps we could reach out to the guy behind QGoodWindow and the FPA could offer him a grant to implement it in FreeCAD?

@MisterMakerNL have you started to work on this? Are you willing to do it?

yorikvanhavre, 2024-03-13:
I would start assessing if QGoodWindow does what we want it to do... Because we don't strictly need it, it's doable without.
I understand the toolbar is actually just a QToolBar with widgets on it ( https://github.com/johanhelsing/qt-csd-demo/blob/master/filebrowser/main.qml ) and a different drag mechanism that it drags the entire window instead of just that toolbar.

@AthidhiVijai that is not the topic that is being discussed here... It's best to open a new issue / feature request otherwise your issue might get overlooked and forgotten

MisterMakerNL, 2024-03-13:
@PaddleStroke I am too bad at coding to get this done in any reasonable time.

AthidhiVijai, 2024-03-14:
okay @yorikvanhavre , thanks for your guidance , will open a new issue

yorikvanhavre, 2024-03-14:
I would roughly imagine something like this: all what's below should be optional and switched by a preference parameter:

* A new QToolBar is created, that should [sit at the very top above the others](https://stackoverflow.com/questions/1441143/how-to-set-the-line-where-a-qtoolbar-is-displayed), take the whole windows width, and have its [DragHandler](https://www.qt.io/blog/custom-window-decorations) set to the main window's [startSystemMove](https://doc-snapshots.qt.io/qt5-5.15/qwindow.html#startSystemMove) slot
* The main window has the `Qt.FramelessWindowHint` flag so the OS does not draw the window frame
* The [menu bar](https://doc.qt.io/qt-6/qmainwindow.html#menuBar) is hidden
* The MDITabbar under the MDI view is hidden
* The main window should also implement `startSystemResize` so one can drag the edges to resize the window (which is normally done by the window frame and controlled by the OS)

Once that is working then we can start moving widgets to the title QToolbar (the workbench switcher, FreeCAD's MDITabbar, etc...) and a minimize/maximize button and a close button and I guess the job is done...

If nobody wants, I can start working on the stuff above, but not before a few weeks I'm afraid. If anyone fancies trying, I can help, though!

yorikvanhavre, 2024-03-15:
Playing a bit with python, but not there yet. We need to be able to insert a toolbarbreak and a toolbar before all the other toolbars.

```python
from PySide import QtGui, QtCore
mw = FreeCADGui.getMainWindow()

# add a new toolbar
titlebar = QtGui.QToolBar()
mw.addToolBar(QtCore.Qt.TopToolBarArea,titlebar)
# here we'd need to insertToolBarBreak() and insertToolBar() instead
# so the titlebar goes above all the others

# add an icon
icon = QtGui.QIcon(":/icons/freecad.svg")
iconlabel = QtGui.QLabel()
iconlabel.setPixmap(icon.pixmap(24,24))

# make the toolbar unmovable
titlebar.show()
titlebar.setObjectName("TitleBar")
titlebar.setMovable(False)

# getting the workbench selector (on my computer it's on the menu bar)
mb = mw.menuBar()
mb.cornerWidget(QtCore.Qt.TopLeftCorner)
#<PySide2.QtWidgets.QComboBox(0x5615f7771c20) at 0x7f976670bfc0>
wbs = mb.cornerWidget(QtCore.Qt.TopLeftCorner)

# hide menu
mw.menuBar().hide()

# get the mdi tabs
cw=mw.centralWidget()
cw.children()
#cw.children()[1]
#<PySide2.QtWidgets.QTabBar(0x5615f5bcd170, name="mdiAreaTabBar") at 0x7f9853eb7a80>
tb = cw.children()[1]
mdi = tb.parent()
mdi.removeSubWindow(tb)

# this does not work very well...
#titlebar.addWidget(iconlabel)
#titlebar.addWidget(wbs)
#titlebar.addWidget(tb)
titlebar.setLayout(QtGui.QHBoxLayout())
titlebar.layout().addWidget(iconlabel)
titlebar.layout().addWidget(wbs)
titlebar.layout().addWidget(tb)

# set geometry and unmovable
titlebar.setMovable(False)
titlebar.setIconSize(QtCore.QSize(16,16))
titlebar.setMinimumHeight(48)
```

PaddleStroke, 2024-03-15:
I'm not sure what you're planing to do is possible. There's no such thing as tool bar breaks. I don't think you can deliminate the space.
I think the correct way of doing this is to : 
- Hide title bar.
- Keep the menu bar.
- Create a 'Toolbar Docking area' widget. For this we reuse what RT did for the status bar.
- Add 'Toolbar Docking area' to the left and right corner widget of the menu bar.
- To the right corner widget we also add the window buttons.
- Add an option to hide the menu itself, not the menubar.

This also have the advantage of being visually different from toolbars.

yorikvanhavre, 2024-03-15:
>There's no such thing as tool bar breaks. I don't think you can deliminate the space.

Here: https://doc.qt.io/qt-6/qmainwindow.html#insertToolBarBreak

Your plan looks good too, but then why not replace the menu bar entirely?

https://doc.qt.io/qt-6/qmainwindow.html#setMenuWidget

PaddleStroke, 2024-03-15:
Ah look at that :D The one time I have not started a sentenced by 'I think' I'm proven wrong ;)

The advantage of using the menubar is that those who want to keep the menu can. But maybe there are other solution.

yorikvanhavre, 2024-03-15:
> Ah look at that :D The one time I have not started a sentenced by 'I think' I'm proven wrong ;)

I couldn't resist in proving you wrong... Easy dopamine

I hadn't thought about that, indeed that could be very flexible... Be able to switch titlebar on/off AND menu on/off (in which case it becomes a hamburger menu).

Firefox/thunderbird also allow to press Alt to reveal the menu... That works well too IMHO

PaddleStroke, 2024-05-13:
@realthunder By chance are you interested to tackle this title-bar question? It could make for a really nice UI improvement : 
[image]

If yes to summarize for you, we have 2 sources : 
- https://www.qt.io/blog/custom-window-decorations : qt basic API (but resize apparantly does not work for macOS).
- https://github.com/antonypro/QGoodWindow : QGoodWindow API : Works on all plateform, and also implement the decorations for all the plateform. So overall this sounds like a more complete and simpler solution. However I could not get it to build within FreeCAD yet. Also QGoodWindow implements a dark/light themes system which is probably going to conflict with FreeCAD's. Though from the screenshots perhaps QGoodWindow's themes are better...

yorikvanhavre, 2024-05-13:
The "drag-to-menubar" commit actually changes a lot of things and adds some possibilities... This whole topic could then boil down to:

* An option to change the menu items into a hamburger menu (which contains the menu items)
* Make the menubar draggable (it then drags the entire window as explained in the qt blog article)
* An option to disable drawing the window decorations
* Make the tabs bar draggable

Then the rest would be for the user to set up

microbearsmb, 2024-05-30:
If this is still being pursued... what yorik mentioned in the last comment would be amazing. Looking at Thunderbird, a customizable/functional title bar looks really clean and saves significant screen space. So if the easiest solution now is to just let users disable the default titlebar, and treat the menu bar like the title bar... that would be great.

luzpaz, 2024-09-08:
This would be helpful for FreeCAD on Raspberry Pi
 
[image]

source: @concretedog on [twitter](https://x.com/concreted0g/status/1832868593685647407)

AdrianCM95, 2024-12-24:
In reference to #18630 

This is an improvement request over the current user interface:

- to remove the window top bar to have more working space.
- to use the Freecad icon as a super key to hide the normal text menus under it (and also display a search bar to quickly find commands like #8290)

In order to call this function, a shortcut like  Superkey+F, Shift+F, (F,F)..., or whatever combination could be used.
I think this can help to speed up the workflow even without knowing all the shortcuts for the rest of the functions.

This is the current layout:
[image]

This is the proposed layout:
[image]

PaddleStroke, 2025-11-18:
Perhaps I can make a PR to upstream with it being disabled by default.

pierreporte, 2025-11-18:
Does it support Linux window managers?

PaddleStroke, 2025-11-18:
Linux support is hit and miss so I don't know. In AstoCAD it works really well on windows. But linux and mac it's not 100% perfect. Hence the 'disabled by default' suggestion

Maybe getting it in core will attract people to make it better for linux and mac

pierreporte, 2025-11-18:
I say go ahead. If it’s empty and disabled by default we will have plenty of time to test it and decide how to fill it.

luzpaz, 2025-12-25:
> Maybe getting it in core will attract people to make it better for linux and mac

@PaddleStroke yes, that makes sense. There are some people looking at this, for example on reddit 
* https://www.reddit.com/r/FreeCAD/comments/1pte2q7/steam_deck_with_freecad_is_pretty_slick/
* https://www.reddit.com/r/FreeCAD/comments/1ppse4y/i_made_a_sliding_screen_rasberry_pi_computer_for/

PaddleStroke, 2025-12-29:
I'll PR later when PR start merging for 1.2

MisterMakerNL, 2026-11-01:
Fixed by downloading and installing Solidcraft.

ttomasini, 2026-11-01:
I'm pretty sure this is the wrong status for this issue.
@yorikvanhavre could you please re-open this ticket?