FreeCAD/FreeCAD #12744: [Problem] v0.22.0dev  STL export produces open edges

7th-fret, 2024-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have this problem a few days now.
Many times when i export a body to stl and open it in prusa-slicer, it reports errors and open edges.

So, i have searched in existing issues and found something similar #11985 , about export with open edges.
This issue was closed by the commit [ad605d6](https://github.com/FreeCAD/FreeCAD/commit/ad605d6e2387deba58c59b3a66354fb38d6ba60a)

As a test, i checkout and compiled the parent commit [df7e783](https://github.com/FreeCAD/FreeCAD/commit/df7e78351340b5528dee6e8caef69a57483b02b4). Then i opened the same FCStd file, i exported a new STL and it was error-free and without open edges.
After this, i made the same process for the commit [ad605d6](https://github.com/FreeCAD/FreeCAD/commit/ad605d6e2387deba58c59b3a66354fb38d6ba60a) to confirm if the problem returns. And yes the errors and open edges returned.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36236 (Git)
Build type: Release
Branch: main
Hash: 0420f72ca0d72198f1eaf347d31479f29658efc4
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.23
  * freecad.gears 1.2.0
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
  * fasteners 0.5.12
```

### Subproject(s) affected?

None

### Anything else?

[OpenEdges.zip](https://github.com/FreeCAD/FreeCAD/files/14480061/OpenEdges.zip)

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-04:
@wwmayer you authored the mentioned commits, maybe you could weigh in here.

wwmayer, 2024-03-04:
I have PrusaSlicer 2.4.2 and it doesn't complain about the exported mesh files. I have tested the STL and 3mf formats.

If I switch to the Mesh wb and use the function to create a mesh from a shape then the Standard method (deviation=0.5mm, angle=28.5 deg) creates a mesh with 724 points and 1464 triangles.
The mesh evaluation function shows that the mesh has six duplicated points. This kind of error can be easily fixed and after clicking the Repair button the mesh is correct.

With v0.21 the meshing function correctly creates a mesh with 718 points and 1464 triangles.

So, if you use the meshing function what kind of errors does the evaluation dialog show?

JohnOCFII, 2024-03-05:
> I have PrusaSlicer 2.4.2 and it doesn't complain about the exported mesh files. I have tested the STL and 3mf formats.

Do you have 2.4.2 or 2.7.2?  2.7.2 was released this past week.  PrusaSlicer 2.4.2 was released in April 2022.

wwmayer, 2024-03-05:
2.4.2. I have downloaded 2.7.2 and it still doesn't complain on import. But when switching in Expert mode I now see a warning that says 20 open edges.

For the time being you can use the STL format as there this error will be fixed automatically. Or if you prefer 3mf you should load it into FreeCAD, repair the defect and re-export.

7th-fret, 2024-03-05:
> If I switch to the Mesh wb and use the function to create a mesh from a shape then the Standard method (deviation=0.5mm, angle=28.5 deg) creates a mesh with 724 points and 1464 triangles. The mesh [...]

The attached demo part OpenEdges.FCStd (first sketch originally designed under YZ_Plane), in mesh wb create mesh from a shape with deviation=0.5mm, angle=28.5 deg it produces this and analyzer reports errors...
[image]

Then i thougth, what if i change the attachment support of the first sketch (Sketch) from YZ_Plane to XY_Plane...
I get this and analyzer reports NO errors...
[image]

wwmayer, 2024-03-05:
I can confirm all the reported observations but I still struggle to understand the root causes.

wwmayer, 2024-03-06:
OK, I found now the reason why this happens.

Internally a `std::set` is used and the Vertex class implements the `<` operator that compares the x,y,z coordinates. It first compares the x values, if they are equal the y values and if they are equal the z values. This guarantees that the points are strictly ordered by their x, y and z values. And it guarantees that every point in the set is unique.

Now by using a certain tolerance (here 2.2e-15) it cannot be guaranteed any more that the points are strictly ordered by their x,y,z values and as a consequence it can happen that not every point is unique any more.

Thus, some points in the mesh data structure may appear twice but the facets then only reference one of the duplicated points. And this causes the mesh topology to have some open edges.

commit eb32abe7 (2024-03-11): Part: fixes #12744: mesh export creates open edges

For the concatenated domains check if points are duplicate and if yes remove them