FreeCAD/FreeCAD #12746: [Problem] Ctrl+A to select all the geometries in a sketch

FEA-eng, 2024-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some time ago @PaddleStroke added a really cool feature that allows the use of standard copy, cut and paste keyboard shortcuts in the Sketcher  (#11537). I think that it would be nice to also have the Ctrl+A shortcut implemented to select all the geometries (and then copy/cut them with another shortcut). Currently, one has to drag the selection box to do that. This could also eliminate the need for this tool: [Sketcher DeleteAllGeometry](https://wiki.freecad.org/Sketcher_DeleteAllGeometry) since it would be just a matter of using Ctrl+A followed by Delete.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Be-Cool-Man, 2024-03-04:
Additionally, there should be a shortcut for selecting all constraints E.G., [CTRL]+[Shift]+[A]

hasecilu, 2024-03-05:
According to https://wiki.freecad.org/Sandbox:Keyboard_Shortcuts shortcuts using `Ctrl` key should be reserved for global commands.

`Ctrl+A` already works on tree view and task panel, specifically for Sketcher, pressing `Ctrl+A` selects all elements but only the element that was originally selected is highlighted in the viewport. Using `Ctrl` to add another element to selection work, using `Shift+Ctrl` to add various elements to selection not work.

[image]

FEA-eng, 2024-03-07:
> `Ctrl+A` already works on tree view and task panel, specifically for Sketcher, pressing `Ctrl+A` selects all elements but only the element that was originally selected is highlighted in the [...]

Right, but it needs a major overhaul. For geometry, it works only when the Elements tab has the focus, doesn't highlight anything else than the current manual selection and doesn't work with the new copy/cut shortcuts.

maxwxyz, 2024-03-09:
> Additionally, there should be a shortcut for selecting all constraints E.G., [CTRL]+[Shift]+[A]

What would be the use case for this? Copy/Pasting constraints do not work. You could select all or multiple in the task dialog to change them to driven. There is also the command to delete all constraints.

closed 2025-08-29 by kadet1090