FreeCAD/FreeCAD #12752: [Problem] Segfault when using the new Assembly

leoheck, 2024-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was experimenting with the new Assembly for the first time. Using the Fixed Joint with the grounded model.

Then when I tried that, the axis of 2 wholes did not align as expected. The axis of one of the parts was in an odd place location. I tried many times to select holes to make this joint and nothing. 

After some tries, on the last try I realized that I could add more parts of the same while adding one part. Then after adding the second of the same, I hit Esc to cancel this, keeping just one part. Then Freecad crashed. 

While working with the new `Assembly` workbench I got some segfaults.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git) AppImage
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.7
  * sheetmetal 0.4.2
  * InventorLoader 1.4.0
  * ose-piping
  * fasteners 0.5.12
  * OpenDark 2023.12.17
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.1.23
  * OpticsWorkbench 1.0.11
  * lattice2 1.0.0
  * kicadStepUpMod 10.23.2
```
```

### Subproject(s) affected?

Assembly

### Anything else?

This is the output of the command line.

```
Latest FreeCAD /home/lheck/Applications/FreeCAD_weekly-builds-36228-2024-03-01-conda-Linux-x86_64-py310.AppImage
FreeCAD 0.22.0, Libs: 0.22.0devR36228 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
Reading failed from embedded file: PartShape22.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: PartShape76.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: PartShape121.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: PartShape189.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: PartShape254.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: PartShape256.bin (0 bytes, 2 bytes compressed)
[... 75 lines cut ...]
#43  0x7f78a45763ab in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#44  /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e39) [0x7f78a45c6e39]
#45  /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Widgets.so.5(+0x1c9f99) [0x7f78a45c9f99]
#46  0x7f78a457035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#47  0x7f78a5f25f89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libFreeCADGui.so+0x99
#48  0x7f78a36b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Core.so.5+0x108
#49  0x7f78a3d34bec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Gui.so.5+0x84c
#50  0x7f78a3d0ad7c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Gui.so.5+0xac
#51  /tmp/.mount_FreeCAeyRpLm/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x6454a) [0x7f789de4c54a]
#52  /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/./libglib-2.0.so.0(+0x5a417) [0x7f78a1d11417]
#53  /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/./libglib-2.0.so.0(+0x5d3c7) [0x7f78a1d143c7]
#54  /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f78a1d149be]
#55  0x7f78a37083d6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Core.so.5+0x66
#56  0x7f78a36b2bbb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Core.so.5+0x11b
#57  0x7f78a36ba6a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libQt5Core.so.5+0x82
#58  0x7f78a5e899c7 in Gui::Application::runApplication() from /tmp/.mount_FreeCAeyRpLm/usr/bin/../lib/libFreeCADGui.so+0xa27
#59  /tmp/.mount_FreeCAeyRpLm/usr/bin/freecad(+0x7620) [0x56121607c620]
#60  /lib/x86_64-linux-gnu/libc.so.6(+0x28150) [0x7f78a3028150]
#61  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x7f78a3028209]
#62  /tmp/.mount_FreeCAeyRpLm/usr/bin/freecad(+0x7981) [0x56121607c981]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-04:
Could you provide an example file or detailed steps how to reproduce, I cannot reproduce. Has it to do with `Esc` while adding new parts or is it related to grounded and fixed parts?

leoheck, 2024-03-04:
I am not sure, but I will try to make this example.

leoheck, 2024-03-04:
The first of my problems was making the alignment of those 2 parts impossible, one of them had as a tip some feature in the middle of the tree. This was fixed, but I still have to try to break Freecad again. Right now I am a bit out of time.

leoheck, 2024-03-05:
Hey @maxwxyz,  I could break Freecad again, I am attaching my file for you to try to reproduce it.

The steps to break it:
1. Activate the Assembly
2. Select 2 circular edges on both linked parts.
3. Apply the Fixed Joint. 

I chose these 2 edges:

[image]

And watch the explosion.

The important part is this. The mountplate has its Tip on a feature that was not the last one (yeah, this is not what I wanted). I forgot to change this when I was fixing this part. Also, the selected edge for the Joint is available on the last feature. 

But these things should not break the tool, right?

See the picture below:

[image]

I feel that there is a problem with the linked part. It should be displaying the Tip of the part no matter where the Tip is, right? But it is showing the whole part. I feel that fixing this would prevent this segfault. Because I would not be able to select an edge that is beyond the object's Tip.

Here is my sample project.
[freecad-sefgault-assembly.zip](https://github.com/FreeCAD/FreeCAD/files/14496254/freecad-sefgault-assembly.zip)

leoheck, 2024-03-06:
I feel that the bug is in the `Link` and not in the `Assembly`, but it could be both.

maxwxyz, 2024-03-06:
I initially got a few errors but the joint worked: 
[image]

Not sure @PaddleStroke how it handles when the tip is not on the last feature.

maxwxyz, 2024-03-06:
To add, the JCS is weirdly placed when hovering and selecting the outer circle:
[image]

leoheck, 2024-03-07:
I think this is very unstable yet. Everything I try to do breaks Freecad.
Here is another one, I am not sure it is a new issue, but the objects are different this time.
Also, I upgraded Freecad yesterday, I think.

For instance, now I am trying to create a RevoluteJoint with the selected edges.
Both parts are Part Objects that were cloned from a single Part, and transformed to be separated things.

[image]

```
➜ freecad-daily 
Latest FreeCAD /home/lheck/Applications/FreeCAD_weekly-builds-36277-2024-03-04-conda-Linux-x86_64-py310.AppImage
FreeCAD 0.22.0, Libs: 0.22.0devR36277 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
Ignore recovery file  "/home/lheck/.cache/FreeCAD/Cache/FreeCAD_Doc_fb707632-3de5-488c-ad89-c0b09b7f4bad_50994f_14149/fc_recovery_file.fcstd"  because it is older than the project file "/home/lheck/Dropbox/deepx/FBI/case/case-v0.2.4.FCStd" 

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42990) [0x713e6c242990]
#1  0x713cc8cfb240 in Assembly::AssemblyObject::fixGroundedPart(App::DocumentObject*, Base::Placement&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) from /tmp/.mount_FreeCAm70ZUv/usr/lib/AssemblyApp.so+0x30
#2  0x713cc8cfbde2 in Assembly::AssemblyObject::fixGroundedParts() from /tmp/.mount_FreeCAm70ZUv/usr/lib/AssemblyApp.so+0x112
#3  0x713cc8d0b53b in Assembly::AssemblyObject::solve(bool) from /tmp/.mount_FreeCAm70ZUv/usr/lib/AssemblyApp.so+0xeb
#4  0x713cc8d0d68e in Assembly::AssemblyObjectPy::solve(_object*) from /tmp/.mount_FreeCAm70ZUv/usr/lib/AssemblyApp.so+0x6e
#5  0x713cc8d0d714 in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) from /tmp/.mount_FreeCAm70ZUv/usr/lib/AssemblyApp.so+0x24
#6  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(+0x149ba8) [0x713e6e949ba8]
#7  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x27b) [0x713e6e942edb]
#8  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x541e) [0x713e6e93ed5e]
#9  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x713e6e94a015]
#10  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x32d) [0x713e6e939c6d]
#11  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x713e6e94a015]
#12  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x739) [0x713e6e93a079]
#13  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(+0x155e32) [0x713e6e955e32]
#14  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4ba0) [0x713e6e93e4e0]
#15  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(+0x155e32) [0x713e6e955e32]
#16  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4ba0) [0x713e6e93e4e0]
#17  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_FastCallDictTstate+0xd7) [0x713e6e9423a7]
#18  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(+0x1538e8) [0x713e6e9538e8]
#19  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x28b) [0x713e6e942eeb]
#20  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4e3a) [0x713e6e93e77a]
#21  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x713e6e94a015]
#22  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x32d) [0x713e6e939c6d]
#23  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libpython3.10.so.1.0(+0x1560a2) [0x713e6e9560a2]
#24  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libFreeCADBase.so(PP_Run_Method+0xe3) [0x713e6dfdd7b3]
#25  0x713e6df28f8b in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libFreeCADBase.so+0x2b
#26  0x713e6f179d13 in Gui::PythonCommand::activated(int) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libFreeCADGui.so+0x53
#27  0x713e6f17a3a8 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libFreeCADGui.so+0x2a8
#28  0x713e6f17a7a4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libFreeCADGui.so+0x134
#29  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libQt5Core.so.5(+0x2e9214) [0x713e6c8e9214]
#30  0x713e6d76a422 in QAction::triggered(bool) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libQt5Widgets.so.5+0x32
#31  0x713e6d76cdeb in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libQt5Widgets.so.5+0xab
#32  /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libQt5Widgets.so.5(+0x25cb9a) [0x713e6d85cb9a]
#33  0x713e6d85ccd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libQt5Widgets.so.5+0xe7
[... cut ...]
#54  0x713e6c8ba6a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libQt5Core.so.5+0x82
#55  0x713e6f08aab7 in Gui::Application::runApplication() from /tmp/.mount_FreeCAm70ZUv/usr/bin/../lib/libFreeCADGui.so+0xa27
#56  /tmp/.mount_FreeCAm70ZUv/usr/bin/freecad(+0x7620) [0x58f43f61b620]
#57  /lib/x86_64-linux-gnu/libc.so.6(+0x28150) [0x713e6c228150]
#58  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x713e6c228209]
#59  /tmp/.mount_FreeCAm70ZUv/usr/bin/freecad(+0x7981) [0x58f43f61b981]
```

PaddleStroke, 2024-03-07:
Regarding last post I need the files and reproducible steps else I can't help

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre