FreeCAD/FreeCAD #12755: [Problem] Regression: Crash in Sketcher Arc

chrisb-github, 2024-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To reproduce:
- create a new sketch
- select the arc creation icon
- select the origin for the center
- move the mouse e.g. to the left upper quadrant
- **double click** instead of the correct single click, this can easily happen inadvertently
-> FreeCAD crashes

FreeCAD didn't crash in 0.21.2 on macOS. Instead it correctly shows errors in report view and incorrectly creates a coincidence constraint with no geometric element.

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git)
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.13
  * sheetmetal 0.4.2
  * ExplodedAssembly
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-05:
Can reproduce on <details><summary>Version</summary>
<p>

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * Curves 0.6.8
  * Defeaturing 1.2.0
  * fasteners 0.4.56
  * OpenDark 2023.10.22
  * sheetmetal 0.2.63

```

</p>
</details> 

Won't crash but this is the error and FC is frozen:

<details><summary>Log</summary>
<p>

```
08:20:48  Sketcher constraint number 1 is malformed!
08:20:48  Unhandled exception in ViewProvider::eventCallback: Geometry element does not have solver information (possibly when trying to apply widget constraints)!
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
08:20:48  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:20:48  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:20:48  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:20:51  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:20:58  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:21:30  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

</p>
</details>

xtemp09, 2024-03-05:
I cannot reproduce it with

<details><summary>Ubuntu 22.04.3 LTS (KDE/plasma)</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36236 (Git)
Build type: Debug
Branch: main
Hash: 0420f72ca0d72198f1eaf347d31479f29658efc4
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

No crash, no unhandled exception takes place, everything just works fine.

wwmayer, 2024-03-05:
Confirmed.

An exception is raised but not handled properly. This later causes a crash.

The files and line number of the function calls when the exception is thrown:
```
FreeCAD/src/Gui/ViewProvider.cpp:256
FreeCAD/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp:956
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchControllableHandler.h:95
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchControllableHandler.h:142
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:619
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:212
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:143
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchControllableHandler.h:164
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:633
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:495
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchControllableHandler.h:147
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchHandlerArc.h:835
FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:979
FreeCAD/src/Base/Exception.cpp:533
```
The error message is: `Geometry element does not have solver information (possibly when trying to apply widget constraints)!`

0penBrain, 2024-05-01:
Just in case it may help, I have a very different trace here:
```
#0 SoAction::traverse() at obj-x86_64-linux-gnu/src/actions/src/actions/SoAction.cpp:943
#1 SoChildList::traverse() at obj-x86_64-linux-gnu/src/misc/src/misc/SoChildList.cpp:357
#2 SoGroup::doAction() at obj-x86_64-linux-gnu/src/nodes/src/nodes/SoGroup.cpp:548
#3 SoSeparator::doAction() at obj-x86_64-linux-gnu/src/nodes/src/nodes/SoSeparator.cpp:487
#4 Gui::SoFCUnifiedSelection::handleEvent() at /home/ob/dev/FreeCAD/src/Gui/SoFCUnifiedSelection.cpp:697
#5 SoAction::traverse() at obj-x86_64-linux-gnu/src/actions/src/actions/SoAction.cpp:950
#6 SoChildList::traverse() at obj-x86_64-linux-gnu/src/misc/src/misc/SoChildList.cpp:357
#7 SoGroup::doAction() at obj-x86_64-linux-gnu/src/nodes/src/nodes/SoGroup.cpp:548
#8 SoSeparator::doAction() at obj-x86_64-linux-gnu/src/nodes/src/nodes/SoSeparator.cpp:487
#9 SoAction::traverse() at obj-x86_64-linux-gnu/src/actions/src/actions/SoAction.cpp:950
#10 SoHandleEventAction::beginTraversal() at obj-x86_64-linux-gnu/src/actions/src/actions/SoHandleEventAction.cpp:376
#11 SoAction::apply() at obj-x86_64-linux-gnu/src/actions/src/actions/SoAction.cpp:565
#12 SoEventManager::actuallyProcessEvent() at obj-x86_64-linux-gnu/src/misc/src/misc/SoEventManager.cpp:366
#13 SIM::Coin3D::Quarter::QuarterWidget::processSoEvent() at /home/ob/dev/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:1 028
#14 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent() at /home/ob/dev/FreeCAD/src/Gui/Quarter/SoQTQuarterAdaptor.cpp:752
#15 Gui::View3DInventorViewer::processSoEventBase() at /home/ob/dev/FreeCAD/src/Gui/View3DInventorViewer.cpp:2 571
#16 Gui::NavigationStyle::processSoEvent() at /home/ob/dev/FreeCAD/src/Gui/NavigationStyle.cpp:1 486
#17 Gui::CADNavigationStyle::processSoEvent() at /home/ob/dev/FreeCAD/src/Gui/CADNavigationStyle.cpp:330
#18 Gui::NavigationStyle::processEvent() at /home/ob/dev/FreeCAD/src/Gui/NavigationStyle.cpp:1 457
#19 Gui::View3DInventorViewer::processSoEvent() at /home/ob/dev/FreeCAD/src/Gui/View3DInventorViewer.cpp:2 566
#20 SIM::Coin3D::Quarter::EventFilter::eventFilter() at /home/ob/dev/FreeCAD/src/Gui/Quarter/EventFilter.cpp:167
#21 QCoreApplicationPrivate::sendThroughObjectEventFilters() at src/corelib/kernel/qcoreapplication.cpp:1 190
#22 QCoreApplicationPrivate::sendThroughObjectEventFilters() at src/corelib/kernel/qcoreapplication.cpp:1 179
#23 QApplicationPrivate::notify_helper() at src/widgets/kernel/qapplication.cpp:3 631
#24 QApplication::notify() at src/widgets/kernel/qapplication.cpp:3 081
#25 Gui::GUIApplication::notify() at /home/ob/dev/FreeCAD/src/Gui/GuiApplication.cpp:83
#26 QCoreApplication::notifyInternal2() at src/corelib/kernel/qcoreapplication.cpp:1 064
#27 QApplicationPrivate::sendMouseEvent() at src/widgets/kernel/qapplication.cpp:2 619
#28 QWidgetWindow::handleMouseEvent() at src/widgets/kernel/qwidgetwindow.cpp:683
#29 QWidgetWindow::event() at src/widgets/kernel/qwidgetwindow.cpp:300
#30 QApplicationPrivate::notify_helper() at src/widgets/kernel/qapplication.cpp:3 637
#31 Gui::GUIApplication::notify() at /home/ob/dev/FreeCAD/src/Gui/GuiApplication.cpp:83
#32 QCoreApplication::notifyInternal2() at src/corelib/kernel/qcoreapplication.cpp:1 064
#33 QGuiApplicationPrivate::processMouseEvent() at src/gui/kernel/qguiapplication.cpp:2 282
#34 QWindowSystemInterface::sendWindowSystemEvents() at src/gui/kernel/qwindowsysteminterface.cpp:1 169
#35 xcbSourceDispatch() at src/plugins/platforms/xcb/src/plugins/platforms/xcb/qxcbeventdispatcher.cpp:105
#36 g_main_context_dispatch() at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#37 ??() at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#38 g_main_context_iteration() at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#39 QEventDispatcherGlib::processEvents() at src/corelib/kernel/qeventdispatcher_glib.cpp:423
#40 QEventLoop::exec() at src/corelib/global/qflags.h:69
#41 QCoreApplication::exec() at src/corelib/global/qflags.h:121
#42 (anonymous namespace)::tryRunEventLoop() at /home/ob/dev/FreeCAD/src/Gui/Application.cpp:2 042
#43 (anonymous namespace)::runEventLoop() at /home/ob/dev/FreeCAD/src/Gui/Application.cpp:2 062
#44 Gui::Application::runApplication() at /home/ob/dev/FreeCAD/src/Gui/Application.cpp:2 139
#45 main() at /home/ob/dev/FreeCAD/src/Main/MainGui.cpp:278
```

0penBrain, 2024-05-02:
@FlachyJoe @PaddleStroke may you a have a clue what happens here ? Hard crash is always a problem due to data loss.

FlachyJoe, 2024-05-03:
#13802 fixes the crash but invalid constraints are created at double-click whatever the geometry tool.

0penBrain, 2024-05-03:
> #13802 fixes the crash but invalid constraints are created at double-click whatever the geometry tool.

Indeed it fixes the crash. :) But still there are potential issues. AFAIK, this is the only geometry tool that allows a double-click (or actually takes it into account). Do you know others ?

FlachyJoe, 2024-05-03:
@0penBrain with an empty sketch try to create a polyline with origin as first point and double-click the second. Ellipse and BSpline by knots are buggy too.

0penBrain, 2024-05-03:
> @0penBrain with an empty sketch try to create a polyline with origin as first point and double-click the second. Ellipse and BSpline by knots are buggy too.

OK. Tried it. This seems like different issues. With arc, you can try the following (with your correction applied):
 * Draw a first arc (normal, any shape) with center constrained on the root point
 * Use one of the previous arc endpoints to constrain center of a second arc, then double-click => this same bug (as described in this issue) makes the first arc to collapse entirely at the root point. It looks like the coincident constraint is "transferred" to the 1st arc (as 2nd hasn't been created).

BTW, would be good to open issues for polyline/ellipse/bspline issues. :wink:

PR #13802 (2024-05-06): Sketcher: Add a guard for not asking information of a point of a curve which doesn't exist
Fix #12755
closed 2024-05-06 by sliptonic