FreeCAD/FreeCAD #12770: [Problem] Cannot edit fixed joints - python exception

furgo16, 2024-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating a simple assembly, I no longer can edit its joints. At some point, this worked by double-clicking on the joints and the task bar would show the editable properties. But after doing this action a couple of times, something happened and they would no longer show up in the task bar. Steps to reproduce:

1. Open attached file => [Box Assembly builtin.zip](https://github.com/FreeCAD/FreeCAD/files/14493932/Box.Assembly.builtin.zip)
2. Load Assembly workbench
3. Click on the Assembly object on the tree to expand it
4. Click on the Joints object under Assembly on the tree to expand it
5. Double-click on the "Fixed" joint to edit it
6. The following exception occurs. Nothing is shown on the task bar

```python traceback
11:26:49  pyException: Traceback (most recent call last):
  File "/snap/freecad/946/usr/Mod/Assembly/JointObject.py", line 873, in doubleClicked
    panel = TaskAssemblyCreateJoint(0, vobj.Object)
  File "/snap/freecad/946/usr/Mod/Assembly/JointObject.py", line 1050, in __init__
    self.updateTaskboxFromJoint()
  File "/snap/freecad/946/usr/Mod/Assembly/JointObject.py", line 1256, in updateTaskboxFromJoint
    obj1 = UtilsAssembly.getObjectInPart(self.joint.Object1, self.joint.Part1)
  File "/snap/freecad/946/usr/Mod/Assembly/UtilsAssembly.py", line 249, in getObjectInPart
    if part.Name == objName:
<class 'AttributeError'>: 'NoneType' object has no attribute 'Name'
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36285 (Git) Snap 946
Build type: Release
Branch: main
Hash: b7a46805040662019c54089a96b3f43caf008e16
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.23
  * ExplodedAssembly
  * BillOfMaterials 0.0.5.6
  * Ondsel-Lens 2024.1.22.01
  * Assembly4 0.50.7
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-03-06:
I've managed to reproduce this situation in another assembly, with the following steps and the accompanying screencast:

1. Expand "Joints" in the tree
2. Double click on one of the joints under "Joins" in the tree. This will show its properties in the Tasks panel.
3. Without clicking on "OK" or "Cancel" in the Tasks panel, double-click on another joints under "Joins" in the tree
4. Watch the assembly explode. Lots of errors on the report panel
5. After clicking undo the assembly is reassembled correctly, but it's no longer possible to double-click on joints, leading to the situation in the issue description

[Boom.webm](https://github.com/FreeCAD/FreeCAD/assets/148809153/8d7aa98f-97fd-4b4b-b723-ca5a1cb62f66)

File to reproduce => 
[Box Assembly builtin parts.zip](https://github.com/FreeCAD/FreeCAD/files/14507339/Box.Assembly.builtin.parts.zip)

PaddleStroke, 2024-03-10:
I can reproduce but I don't know how to solve this problem.

I tried to add 

```
    def doubleClicked(self, vobj):
        if Gui.Control.activeDialog():
            Gui.Control.closeDialog()
```
at the beginning of doubleClicked of joint object, but there is still lot of errors.
It appears that my taskbox reject function is not called automatically when it's dismissed with Gui.Control.closeDialog()

@yorikvanhavre do you know how this situation is handled? How to make sure the taskbox reject is called when another command is creating a new taskbox?
Thanks

PaddleStroke, 2024-03-13:
@wwmayer do you please know if it is possible to connect the dismissal of a task dialog to the dialog's class's reject function?
The issue described here happens because the user is launching a new task while a joint edition is under way. This result in the taskbox being removed without its reject function being called.

Thanks

PaddleStroke, 2024-03-14:
Thanks Werner!

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre