FreeCAD/FreeCAD #12771: [Problem] Allow entering expressions for joint properties   on the Tasks bar

furgo16, 2024-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See attached screenshot. When entering joint properties (Offset, Rotation on the screenshot), it's not possible to use expressions. For consistency and to improve parametric design, it should be possible to type an expression there as well.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36285 (Git) Snap 946
Build type: Release
Branch: main
Hash: b7a46805040662019c54089a96b3f43caf008e16
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.23
  * ExplodedAssembly
  * BillOfMaterials 0.0.5.6
  * Ondsel-Lens 2024.1.22.01
  * Assembly4 0.50.7
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-21:
@PaddleStroke is this a lot of work to implement ?

PaddleStroke, 2024-10-22:
I actually don't know. It may just be a 1 line change. I'll check this

PaddleStroke, 2024-10-22:
So actually for offset and rotation it's going to be difficult, because these spinboxes are not direct representation of the joint's property. The joint actually has 2 Placement properties 'Offset1' and 'Offset2' (pr https://github.com/FreeCAD/FreeCAD/pull/16317 is going to change the UI to allow more precise offsets).

Changing the offset and rotation spinboxes is actually changing the Z axis and yaw of the Offset2 placement property. So there's no direct way to bind the spinbox.

However we can apparantly bind easily for the spinboxes that are direct representation of properties : distance, radius and limits. (according to https://forum.freecad.org/viewtopic.php?t=69334)

PaddleStroke, 2024-10-22:
Hoho look at that snippet I found ! 

```
        Gui.ExpressionBinding(self.form.attachmentOffsetX).bind(self.obj,"AttachmentOffset.Base.x")
        Gui.ExpressionBinding(self.form.attachmentOffsetY).bind(self.obj,"AttachmentOffset.Base.y")
        Gui.ExpressionBinding(self.form.attachmentOffsetZ).bind(self.obj,"AttachmentOffset.Base.z")

        Gui.ExpressionBinding(self.form.attachmentOffsetYaw).bind(self.obj,"AttachmentOffset.Rotation.Yaw")
        Gui.ExpressionBinding(self.form.attachmentOffsetPitch).bind(self.obj,"AttachmentOffset.Rotation.Pitch")
        Gui.ExpressionBinding(self.form.attachmentOffsetRoll).bind(self.obj,"AttachmentOffset.Rotation.Roll")
```

We can actually bind to any component of the placement property. So offset and rotation can do it afterall

luzpaz, 2024-10-23:
Congrats paddle! Thanks for all your awesome work!

closed 2024-11-04 by yorikvanhavre