FreeCAD/FreeCAD #12784: [Problem] Meshing coordinates change on minor version update

evelyn-king, 2024-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On trying to move to a new freecad version from 0.21.1 (specifically to v0.21.2), I encountered a strange bug in the Mesh module. It looks like there was a change from centered coordinates to corner coordinates. I made a small repository that reproduces the bug here: https://github.com/evelyn-king/freecad-bug-repro-01.

### Full version info

```shell
I'm running a headless install. Here's the output of freecad.app.Version():

['0', '21', '2', '33771 (Git)', 'https://github.com/FreeCAD/FreeCAD', '2023/11/14 02:33:39', '(HEAD detached at 0.21.2)', 'b9bfa5c5507506e4515816414cd27f4851d00489']
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-06:
Could you try this in the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-03-08:
@evelyn-king JFYI, in the future it's better for triage and testers if you paste the [About](https://wiki.freecad.org/About) info from the About dialog. TIA!

wwmayer, 2024-03-13:
I guess it's the same as in https://github.com/FreeCAD/FreeCAD/issues/12903

maxwxyz, 2024-03-13:
@evelyn-king can you confirm?

wwmayer, 2024-03-13:
I have tested the script bug_reproduction.py with the latest changes for v0.21 and it says:
`Test passed`
After undoing my fix the script says:
```
Test failed
testBB=(-1.0, 1.0, -2.0, 2.0, -3.0, 3.0)
meshBB=(-2.0, 0.0, -4.0, 0.0, -6.0, 0.0)
Traceback (most recent call last):
  File "/home/user/Projects/rel-FreeCAD-0-21/build/Ext/bug_reproduction.py", line 223, in <module>
    test_exportMeshed()
  File "/home/user/Projects/rel-FreeCAD-0-21/build/Ext/bug_reproduction.py", line 211, in test_exportMeshed
    assert testBB == meshBB
AssertionError
```

So, it's indeed the same issue.

maxwxyz, 2024-03-13:
So also closed with https://github.com/FreeCAD/FreeCAD/pull/11539 ?

wwmayer, 2024-03-13:
Yes. The specific commit is: 3aa6e37bb2c76e

closed 2024-03-13 by maxwxyz