FreeCAD/FreeCAD #12785: [Problem] Copy expression from expression editor

Shkolik, 2024-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempt to copy expression from expression editor leads to crash

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git)
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapesWorkbench 1.0.5
  * CurvesWB 0.6.21
  * fasteners 0.4.75
  * Foamcut 0.1.1
  * Marz 0.0.27-alpha
  * PieMenu 1.3.9
```
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/26121979/6164acf5-5816-44ee-a2f9-c8ab1034b32d

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-03-06:
I've copied formulae from and to the expression editor many times in different versions of FreeCAD on Linux and Windows and never encountered a crash or any problem at all.

Could you provide additional information about the crash, like backtrace and system information? You have locale set to _English/United States (en_US)_, but what is your system locale? What is your decimal separator? Does the stable version of FreeCAD work fine? Do you always encounter crash or it crashes only when copying 50.15 mm? If you copy round numbers, does it crash?

Shkolik, 2024-03-06:
`Could you provide additional information about the crash, like backtrace` 

How I can get this information?

`You have locale set to English/United States (en_US), but what is your system locale?`

same English/United States (en_US)

`What is your decimal separator?`

'.' - point, default to en_us

`Does the stable version of FreeCAD work fine?`

Just checked with 0.21.2 - same behavior

`Do you always encounter crash or it crashes only when copying 50.15 mm? If you copy round numbers, does it crash?`

I'm not copying numbers, I trying to copy expression. Basically to crash FreeCAD just open expression editor from any place, type something in there, select typed text and do "CTRL+C" or right mouse button click -> copy.

xtemp09, 2024-03-06:
> How I can get this information?

Only with a debugger, like lldb, gdb or cdb.

> I'm not copying numbers, I trying to copy expression. Basically to crash FreeCAD just open expression editor from any place, type something in there, select typed text and do "CTRL+C" or right mouse button click -> copy.

1. If you just type something, does it crash?
2. If you paste something into the edit field, does it crash?

Shkolik, 2024-03-06:
`Only with a debugger, like lldb, gdb or cdb.`

it's out of my scope then

`If you just type something, does it crash?
If you paste something into the edit field, does it crash?`

No and No

Shkolik, 2024-03-06:
To be clear - it's working fine in a Link branch:

[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 2024.123.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.11.7, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapesWorkbench 1.0.5
  * CurvesWB 0.6.21
  * fasteners 0.4.75
  * Foamcut 0.1.1
  * Marz 0.0.27-alpha
  * PieMenu 1.3.9
[/code]

Shkolik, 2024-03-06:
Tested on different versions of FreeCAD - bug exists even on 0.21.0.

It appears that "CTRL+C" is working fine - only "right mouse button click -> copy" cause a crash.

maxwxyz, 2024-03-06:
I cannot reproduce on 
<details><summary>Weekly Windows 10</summary>
<p>

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * Curves 0.6.8
  * Defeaturing 1.2.0
  * fasteners 0.4.56
  * OpenDark 2023.10.22
  * sheetmetal 0.2.63

``` 

</p>
</details> 

Please provide a file and exact steps how this could be reproduced.

Shkolik, 2024-03-06:
Checked on another machine running Win10 and it works just fine, so looks like issue isolated to Win11. Do not have another Win11 machine to test...

Video with steps to repro:

https://github.com/FreeCAD/FreeCAD/assets/26121979/39692561-4965-4e3c-a062-566e258ca087

File used in video:

[copy_constraint.zip](https://github.com/FreeCAD/FreeCAD/files/14512754/copy_constraint.zip)

maxwxyz, 2024-03-06:
I cannot reproduce on 
<details><summary>Windows 11 dev</summary>
<p>

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2

</p>
</details> 
A fresh file and your file works perfectly fine.
Please try with the latest weekly build https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds as it cannot be reproduced.

Shkolik, 2024-03-06:
Tried with:

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapesWorkbench 1.0.5
  * CurvesWB 0.6.21
  * fasteners 0.4.75
  * Foamcut 0.1.1
  * Marz 0.0.27-alpha
  * PieMenu 1.3.9
```

Same thing - get crash on my machine. Can I run FreeCAD with some arguments but without debugger to collect log data?

Shkolik, 2024-03-06:
Was able to get access to another Win11 machine - checked last weekly build and it's working fine. For me the only difference in behavior is that when I select Copy from menu on my machine - expression editor window closes ant after 1sec I get a crash. On another machine editor window stays open with text selected. So it looks like something steal mouse focus from window on my machine, but have no idea what it is. Checked without externals WB and macros - same thing.

adrianinsaval, 2024-03-07:
I can't reproduce on windows 11 either, there's something wrong in your system.

maxwxyz, 2024-03-07:
@Shkolik could you try this with a fresh install? If no one can reproduce this in a week I will close it otherwise.

Shkolik, 2024-03-07:
> @Shkolik could you try this with a fresh install? If no one can reproduce this in a week I will close it otherwise.

I did and have same issue. Do you have any idea how I can collect some log/crash info without building project myself?

Shkolik, 2024-03-07:
I checked again on another machine and noticed that there expression editor has actual border and on mine it's transparent and has no border at all.
So, I deleted all config files from AppData\Roaming\FreeCAD\ and now it seems to be working fine. Still do not know what was the cause of the problem tho. Can attach my configs if someone is interested.

Shkolik, 2024-03-07:
Looks like this part of user.cfg was causing problem. Have no idea where it came - maybe some artifact from Linkstage3 branch or I just copied config from linkstage, nut without this part everything is working fine.

xtemp09, 2024-03-08:
Hooray! I've OCRed the text you provided, copied and pasted it into the `user.cfg` and _FreeCAD_ now crashes!

Moreover, it crashes even on _Linux_! Nyaha!

It's a typical deallocation of freed memory.

xtemp09, 2024-03-08:
@luzpaz, is it worth making pull request preventing _FreeCAD_ from crashing in this case?

The user severely messed up config, so I don't know if other users can make the same mistake.

adrianinsaval, 2024-03-08:
> @luzpaz, is it worth making pull request preventing _FreeCAD_ from crashing in this case?

IMO yes, did you find out the cause of the crash?

xtemp09, 2024-03-08:
Looks like. The cause of the segmentation fault is access to the already deleted _QMenu_. Simple setting attribute `Qt::WA_DeleteOnClose` and removal of `delete menu` prevented it.

PR #12811 (2024-03-09): Fix the crash described in #12785
Closes #12785. The cause of the segmentation fault is access to the already deleted _QMenu_. Simple setting attribute `Qt::WA_DeleteOnClose` and removal of `delete menu` prevented it. To reproduce the crash, one needs to mess up the `user.cfg` file in the way described in the issue.

Here is the fragment from the issue to mess up the `user.cfg` file:

```xml
<FCParamGroup Name="Expression">
<FCText Name="EditorTrigger">=</FCText>
<FCBool Name="AntoHideEditorIcon" Value="1"/>
<FCBool Name="NoSystemBackground" Value="1"/>
<FCInt Name="EditDialogBGAlpha" Value="1 [... cut]
commit a3036d16 (2024-03-09): Fix the crash described in #12785

Closes #12785. The cause of the segmentation fault is access to the
already deleted _QMenu_. Simple setting attribute `Qt::WA_DeleteOnClose`
and removal of `delete menu` prevented it. To reproduce the crash, one
needs to mess up the `user.cfg` file in the way described in the issue.

Here is the fragment from the issue to mess up the `user.cfg` file:

```xml
<FCParamGroup Name="Expression">
<FCText Name="EditorTrigger">=</FCText>
<FCBool Name="AntoHideEditorIcon" Value="1"/>
<FCBool Name="NoSystemBackground" Value="1"/>
<FCInt Name="EditDialogBGAlpha" Value=" [... cut]
closed 2024-03-09 by wwmayer