FreeCAD/FreeCAD #12789: [Problem] Renaming a grounded part by removing the "lock" character breaks the assembly

furgo16, 2024-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding a part as grounded, it gets a name with a "lock" (🔒) character appended. If that character is removed, the assembly is broken.

Steps to reproduce:
1. Renamed the part in the properties window, and remove the lock character
2. Try to add a fixed joint => there is initially an error shown on the report panel:
   ```python traceback
   10:47:46  Running the Python command 'Assembly_CreateJointFixed' failed:
   Traceback (most recent call last):
     File "/snap/freecad/946/usr/Mod/Assembly/CommandCreateJoint.py", line 92, in Activated activateJoint(0)
     File "/snap/freecad/946/usr/Mod/Assembly/CommandCreateJoint.py", line 54, in activateJoint Gui.Control.showDialog(panel)

   Active task dialog found
   ```
3. If then <kbd>Undo</kbd> is clicked to restore the name back, lots of new errors ensue:
   [image]
   Shortened traceback here, but there are about a thousand instances of this repeated message:
   ```python traceback
   Active task dialog found<class 'RuntimeError'>
   10:48:47  Traceback (most recent call last):
   10:48:47    File "/snap/freecad/946/usr/Mod/Assembly/JointObject.py", line 1376, in moveMouse
   10:48:47      self.joint.ViewObject.Proxy.showPreviewJCS(False)
   10:48:47  RuntimeError: Object has been removed from document
   10:48:47  <class 'RuntimeError'>
   ```

There seems to be some logic relying on the name of the grounded part. If that is the case, the assumption that the name will not be changed by users seems rather unsafe.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36285 (Git) Snap 946
Build type: Release
Branch: main
Hash: b7a46805040662019c54089a96b3f43caf008e16
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.23
  * ExplodedAssembly
  * BillOfMaterials 0.0.5.6
  * Ondsel-Lens 2024.1.22.01
  * Assembly4 0.50.7
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-06:
Related: 
- https://github.com/FreeCAD/FreeCAD/issues/12788
- https://github.com/FreeCAD/FreeCAD/issues/12546

@PaddleStroke this lock icon as part of the name seems to be conflicting.
@obelisk79 DWG: would there be a better option to display grounded parts an an assembly?

obelisk79, 2024-03-06:
It seems that moving that lock indicator to be a true icon rather than part of the name/label would resolve much of the unwanted behavior.

pierreporte, 2024-03-06:
Related to #12141? This is partially about not using the displayed text as the name of the part and replacing it with an attribute modified elsewhere.

furgo16, 2024-03-06:
> Related to https://github.com/FreeCAD/FreeCAD/issues/12141?

After reading, I don't think the two issues are related, other than they both refer to a component's name. Thanks everyone!

adrianinsaval, 2024-03-07:
I think we need to start thinking about displaying multiple icons for an item in the tree. Overlays are too crowded and hard to understand

obelisk79, 2024-03-07:
> I think we need to start thinking about displaying multiple icons for an item in the tree. Overlays are too crowded and hard to understand

I'm not sure what you are trying to say, would you mind explaining what you mean so I can understand? Here or on discord?

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre