FreeCAD/FreeCAD #12793: [Problem] Assembly - Datum element are not taken in account in Assembly WB

Nemesis81, 2024-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It looks like the datum point and other datum feature are not well handle by the Assembly Solver.
test in the video shows 3 assembly of 2 cubes. : 

1. 2 bodies Distance constraint between Vertex and surface => Works OK
2. 2 parts container with points from draftWB and cube from PD => OK
3. 2 bodies with datum points and datum plane NOK
[Video](https://youtu.be/u2SSCj8Ux_g)

Tested on Ondsel version

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git) AppImage
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * DVA_WorkBench
  * fasteners 0.5.12
  * Ondsel-Lens 2024.1.22.01
  * OpenDark 2023.12.17
```
```

### Subproject(s) affected?

Assembly

### Anything else?

maybe related to #11607 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@PaddleStroke

PaddleStroke, 2024-03-21:
I have just tested and datum elements in body should work. I cannot reproduce the problem. See below two bodies, one with a datum point and the other with a datum plane. Distance constraint between the point and plane. Works ok

[image]

Please provide file where I can reproduce.

PaddleStroke, 2024-03-21:
It is to be noted that I tried on my dev branch https://github.com/FreeCAD/FreeCAD/pull/12406.
Though datum support was added before so that may not be relevant. But perhaps it was broken in main and it got fixed in my branch.

luzpaz, 2024-08-18:
What's the verdict here ?

PaddleStroke, 2024-08-28:
I have tested and still can't reproduce. I think this is fixed.

MiqCG, 2024-10-12:
I think maybe this issue can be reopened/reused. I have found problems attaching a JCS to an sketch or to a datum point in revision 1.1.0-38898, but it works correctly in RC1 and RC2.

[Datum_Assembly.zip](https://github.com/user-attachments/files/17350434/Datum_Assembly.zip)

In the attached test files, the auxiliary sketch and the datum point belongs to bbb part (the orange one). You can see that when clicking the first geometry to position the first JCS, it gets to another position.

https://github.com/user-attachments/assets/59a7c8d0-ba3b-4b7e-add3-bcb2ddadf4ba

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

MiqCG, 2024-10-12:
PS: it also works well in revision 38871 (Git) AppImage

luzpaz, 2024-10-12:
I've re-opened this @PaddleStroke

MiqCG, 2024-10-12:
I have just downloaded Appimage revision 38946 and deleted .config .local .cache folders to test again. Same problem positioning JCS as with rev. 38898.

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git) AppImage
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

PaddleStroke, 2024-10-16:
I think the described issue is a regression due to https://github.com/FreeCAD/FreeCAD/pull/16726
If so it will be fixed by https://github.com/FreeCAD/FreeCAD/pull/17267

closed 2024-10-18 by chennes