FreeCAD/FreeCAD #12795: [Problem] Sketcher: (Named) Dimension cannot be used as reference in expressions

maxwxyz, 2024-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to create a dimension constraint and reuse the same by referencing to it inside the same sketch or outside of the sketch with the expression editor.
You have to manually add a new parameter or use an alias in a spreadsheet.
It should be possible to use dimensions (named and unnamed) in FreeCAD as references in the expression editor.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Sketcher

### Anything else?

Related to: 

- https://github.com/FreeCAD/FreeCAD/issues/12145
- https://github.com/FreeCAD/FreeCAD/issues/11893

Maybe related to the new VarSet idea @pieterhijma https://github.com/FreeCAD/FreeCAD/issues/12120

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-03-06:
It's still available
[image]
[image]

maxwxyz, 2024-03-07:
@FlachyJoe thanks for pointing this out, as you always have to type "Constraints." first.
Then its just hard to find, I've created: https://github.com/FreeCAD/FreeCAD/issues/12796

Also, only when looking for references unser `Constraints.` the pulldown menu is not fully visible:
[image]

Referencing Spreadsheets is working fine.

closed 2024-03-07 by maxwxyz

pieterhijma, 2024-03-11:
@maxwxyz, Thank you for pointing that out. Indeed, Sketcher has some support for it, but I would like to generalize that to any dialog that allows you to add an expression. I'm currently investigating these kinds of options and I will make separate issue if I have the problem clear.