FreeCAD/FreeCAD #12798: [Problem] Macro not recording mesh decimation

TheTesla, 2024-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If mesh decimation is used while a macro is recorded, the decimation step is missing in the macro.

Test case:

1.  start macro recording
2. open a stl file
3. select the mesh, which was create from the stl file
4. select "Mesh" mode in the dropdown menu
5. click Mesh_Decimating symbol
6. set parameters to 90% reduction and 0.1 tolerance
7. click ok
8. select "Part" mode
9. select the mesh
10. click "Create shape from mesh..."
11. click ok
12. select the new part
13. click "File" -> "Export..."
14. select STP and save

Result:

- Decimation step is missing, the output file is much larger than expected.

The macro:

```python
# -*- coding: utf-8 -*-

# Macro Begin: C:\Users\efame275\AppData\Roaming\FreeCAD\Macro\convert2.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import Mesh
import Part
import ImportGui

# Gui.runCommand('Std_DlgMacroRecord',0)
### Begin command Std_Open
Mesh.open(u"C:/Users/efame275/Downloads/C-2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u.stl")
# App.setActiveDocument("Unnamed")
# App.ActiveDocument=App.getDocument("Unnamed")
# Gui.ActiveDocument=Gui.getDocument("Unnamed")
# Gui.SendMsgToActiveView("ViewFit")
### End command Std_Open
# Gui.runCommand('Std_OrthographicCamera',1)
# Gui.Selection.addSelection('Unnamed','C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u','',-1.62801,-0.945299,49.875)
### Begin command Std_Workbench
# Gui.activateWorkbench("MeshWorkbench")
### End command Std_Workbench
# Gui.runCommand('Mesh_Decimating',0)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u','',-0.389471,-1.78042,49.875)
### Begin command Std_Workbench
# Gui.activateWorkbench("PartWorkbench")
### End command Std_Workbench
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u','',5.28562,-6.23147,49.0649)
### Begin command Part_ShapeFromMesh
App.getDocument('Unnamed').addObject('Part::Feature', 'C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001')
__shape__ = Part.Shape()
__shape__.makeShapeFromMesh(FreeCAD.getDocument('Unnamed').getObject('C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u').Mesh.Topology, 0.100000, False)
FreeCAD.getDocument('Unnamed').getObject('C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001').Shape = __shape__
FreeCAD.getDocument('Unnamed').getObject('C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001').purgeTouched()
del __shape__
### End command Part_ShapeFromMesh
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001')
### Begin command Std_Export
__objs__ = []
__objs__.append(FreeCAD.getDocument("Unnamed").getObject("C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001"))
if hasattr(ImportGui, "exportOptions"):
    options = ImportGui.exportOptions(u"C:/Users/efame275/Downloads/Unnamed-C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001.step")
    ImportGui.export(__objs__, u"C:/Users/efame275/Downloads/Unnamed-C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001.step", options)
else:
    ImportGui.export(__objs__, u"C:/Users/efame275/Downloads/Unnamed-C_2P8W730680A_screw4knrl_35_trd0600pm_rg9725u_p0200u001.step")

del __objs__
### End command Std_Export
# Macro End: C:\Users\efame275\AppData\Roaming\FreeCAD\Macro\convert2.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++

```

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-12:
Could you test this in the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

TheTesla, 2024-03-12:
> Could you test this in the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Same problem:

```python
# -*- coding: utf-8 -*-

# Macro Begin: /home/stefan/.local/share/FreeCAD/Macro/debugmacro.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import Mesh
import Part
import ImportGui

# Gui.runCommand('Std_DlgMacroRecord',0)
Mesh.insert(u"/home/stefan/projects/SpiroBrixx/CADfiles/screw4knrl_35_trd0500pm_rg9725u_p0200u.stl","Unbenannt")
# App.setActiveDocument("Unbenannt")
# App.ActiveDocument=App.getDocument("Unbenannt")
# Gui.ActiveDocument=Gui.getDocument("Unbenannt")
# Gui.runCommand('Std_OrthographicCamera',1)
# Gui.Selection.addSelection('Unbenannt','screw4knrl_35_trd0500pm_rg9725u_p0200u','',0.0846346,-0.592489,49.875)
### Begin command Std_Workbench
# Gui.activateWorkbench("MeshWorkbench")
### End command Std_Workbench
# Gui.runCommand('Mesh_Decimating',0)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unbenannt','screw4knrl_35_trd0500pm_rg9725u_p0200u','',-1.10033,4.82453,49.8746)
### Begin command Std_Workbench
# Gui.activateWorkbench("PartWorkbench")
### End command Std_Workbench
### Begin command Part_ShapeFromMesh
App.getDocument('Unbenannt').addObject('Part::Feature', 'screw4knrl_35_trd0500pm_rg9725u_p0200u001')
__shape__ = Part.Shape()
__shape__.makeShapeFromMesh(FreeCAD.getDocument('Unbenannt').getObject('screw4knrl_35_trd0500pm_rg9725u_p0200u').Mesh.Topology, 0.100000, False)
FreeCAD.getDocument('Unbenannt').getObject('screw4knrl_35_trd0500pm_rg9725u_p0200u001').Shape = __shape__
FreeCAD.getDocument('Unbenannt').getObject('screw4knrl_35_trd0500pm_rg9725u_p0200u001').purgeTouched()
del __shape__
### End command Part_ShapeFromMesh
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unbenannt','screw4knrl_35_trd0500pm_rg9725u_p0200u001')
### Begin command Std_Export
__objs__ = []
__objs__.append(FreeCAD.getDocument("Unbenannt").getObject("screw4knrl_35_trd0500pm_rg9725u_p0200u001"))
if hasattr(ImportGui, "exportOptions"):
    options = ImportGui.exportOptions(u"/home/stefan/projects/SpiroBrixx/CADfiles/macrodebug.step")
    ImportGui.export(__objs__, u"/home/stefan/projects/SpiroBrixx/CADfiles/macrodebug.step", options)
else:
    ImportGui.export(__objs__, u"/home/stefan/projects/SpiroBrixx/CADfiles/macrodebug.step")

del __objs__
### End command Std_Export
# Macro End: /home/stefan/.local/share/FreeCAD/Macro/debugmacro.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
```

commit af85689c (2024-03-16): Mesh: fixes #12798: Macro not recording mesh decimation
closed 2024-03-16 by wwmayer