FreeCAD/FreeCAD #12800: [Problem] Sketcher GCS: Arc has too many parameters

FlachyJoe, 2024-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Arc parameters are set by

https://github.com/FreeCAD/FreeCAD/blob/2e965999034218b4dc6cf63a0f51e35966182046/src/Mod/Sketcher/App/planegcs/Geo.cpp#L235-L252

and 

https://github.com/FreeCAD/FreeCAD/blob/2e965999034218b4dc6cf63a0f51e35966182046/src/Mod/Sketcher/App/planegcs/Geo.cpp#L208C1-L218C2

This sets 9 parameters for a 5 DOF objects it causes blockage on sketch solver.

`start` and `end` has to be calculated from `center` and `startangle` and `endangle` if they are needed.

Ping sketcher masters: @abdullahtahiriyo @AjinkyaDahale @PaddleStroke 

### Full version info

```shell
FreeCAD 0.22.0, Libs: 0.22.0devR36323
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-03-07:
> Ping sketcher masters: @abdullahtahiriyo @AjinkyaDahale @PaddleStroke

Abdullah has left FreeCAD development: https://forum.freecad.org/viewtopic.php?t=83716

PaddleStroke, 2024-03-07:
I do not know this part of sketcher so I can't give a valuable opinion on this. I think you and Ajinkya knows more about it.

FlachyJoe, 2024-03-08:
Found this
https://github.com/FreeCAD/FreeCAD/blob/eb5a1b333946ce4ac8a4ead835cba8bdcf602a87/src/Mod/Sketcher/App/Sketch.cpp#L944-L948

So the Arc implementation works only in combination with an ArcRules constraint. 
We may add a comment somewhere to alert developers about this necessity at GCS::Arc creation.

AjinkyaDahale, 2024-03-08:
> Found this
>
> https://github.com/FreeCAD/FreeCAD/blob/eb5a1b333946ce4ac8a4ead835cba8bdcf602a87/src/Mod/Sketcher/App/Sketch.cpp#L944-L948
>
> So the Arc implementation works only in combination with an [...]

Where is `GCS::Arc` created which causes this problem? AFAICT they are not supposed to be used in FC outside of Sketch (not SketchObject).

AjinkyaDahale, 2024-05-10:
> EDIT: I'm adding this comment about start and end points in the ArcLength PR
> 
> ```c++
> ...
> ```

@FlachyJoe does this comment complete this issue? Or do you suggest some more active steps?

FlachyJoe, 2024-05-12:
Close #12800 as fixed by #12602

closed 2024-05-12 by FlachyJoe