FreeCAD/FreeCAD #12801: [Feature Request] Open In Finder on macOS

greggjaskiewicz, 2024-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On macOS, usually every "document" driven application - and certainly FreeCAD should be classified to be in this category - has an option to either show the path the current file is in, by hovering or clicking on document icon - FreeCAD does not implement that level of integration with OS. 
Second option that is super useful - is to open the document in Finder (The file system browser). 
This helps, when trying to locate the directory, etc and once again - FreeCAD is missing this. 
[image]

This context menu when right-clicking on the document - should contain a link to "Reveal in Finder" , or such. I'm sure QT has provisions for this built in, and this is a very much expected Pro feature of any document apps on macOS. 

[image]

### Full version info

```shell
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36135 (Git)
Build type: Release
Branch: main
Hash: e99220daf62069be653f73e720bcad9e0cd2fb97
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * A2plus 0.4.62
  * sheetmetal 0.4.0
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.23
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

iAmVaeVictus, 2024-03-07:
Agree. Every improvement of context menu options is a good thing, in my user (of MacOS) opinion. I think there are a lot of "context" right click choices that are absent that should be there, but we need to justify and specify each one to the developers to ensure they are implementing the things that people actually "need".  And, I don't currently understand FC well enough yet to know what's a reasonable ask for context sensitivity vs not in the complex world that is the Workbench-centric FreeCAD, with all it's inherent complexities.

maxwxyz, 2024-03-07:
I think this would also benefit other OS.

greggjaskiewicz, 2024-03-07:
100%

luzpaz, 2024-03-08:
@greggjaskiewicz do you want to take a stab at this ?

greggjaskiewicz, 2024-03-20:
Happy to work on it myself, in Qt there's a code :
void FileUtils::showInGraphicalShell(QWidget *parent, const QString &pathIn)
{
    const QFileInfo fileInfo(pathIn);
    // Mac, Windows support folder or file.
    if (HostOsInfo::isWindowsHost()) {
        const FileName explorer = Environment::systemEnvironment().searchInPath(QLatin1String("explorer.exe"));
        if (explorer.isEmpty()) {
            QMessageBox::warning(parent,
                                 QApplication::translate("Core::Internal",
                                                         "Launching Windows Explorer Failed"),
                                 QApplication::translate("Core::Internal",
                                                         "Could not find explorer.exe in path to launch Windows Explorer."));
            return;
        }
        QStringList param;
        if (!fileInfo.isDir())
            param += QLatin1String("/select,");
        param += QDir::toNativeSeparators(fileInfo.canonicalFilePath());
        QProcess::startDetached(explorer.toString(), param);
    } else if (HostOsInfo::isMacHost()) {
        QStringList scriptArgs;
        scriptArgs << QLatin1String("-e")
                   << QString::fromLatin1("tell application \"Finder\" to reveal POSIX file \"%1\"")
                                         .arg(fileInfo.canonicalFilePath());
        QProcess::execute(QLatin1String("/usr/bin/osascript"), scriptArgs);
        scriptArgs.clear();
        scriptArgs << QLatin1String("-e")
                   << QLatin1String("tell application \"Finder\" to activate");
        QProcess::execute(QLatin1String("/usr/bin/osascript"), scriptArgs);
    } else {
        // we cannot select a file here, because no file browser really supports it...
        const QString folder = fileInfo.isDir() ? fileInfo.absoluteFilePath() : fileInfo.filePath();
        const QString app = UnixUtils::fileBrowser(ICore::settings());
        QProcess browserProc;
        const QString browserArgs = UnixUtils::substituteFileBrowserParameters(app, folder);
        bool success = browserProc.startDetached(browserArgs);
        const QString error = QString::fromLocal8Bit(browserProc.readAllStandardError());
        success = success && error.isEmpty();
        if (!success)
            showGraphicalShellError(parent, app, error);
    }
}

Refactoring and shoving it somewhere in the Tree.cpp actions, would probably work. I'm happy to look at it myself, but might take me a bit to get there :) Never done FreeCAD dev before and I can only compile it on my linux laptop - macOS compilation doesn't seem to be made easier by cmake. .. Basically looks like little love was given to macOS support, which is a shame.

luzpaz, 2024-03-21:
Thanks!

> Basically looks like little love was given to macOS support, which is a shame.  

Just like many other libregraphic foss applications, macOS development lags behind. Probably due to the OS's lack of hospitality toward FLOSS in general.

greggjaskiewicz, 2024-03-27:
@luzpaz exploring how to do it currently. Wish more people were on discord to talk. :)

greggjaskiewicz, 2024-03-30:
I'm having this problem https://forum.freecad.org/viewtopic.php?p=749556#p749556 when building, which blocks me from staring working on this @luzpaz @maxwxyz

luzpaz, 2024-03-30:
Forum thread shows it's not a FC related issue. Did you end up building successfully?

greggjaskiewicz, 2024-03-31:
yes, I got it building fine now - required me to build my own yaml-cpp from scratch etc.

luzpaz, 2024-04-01:
Is the yaml issue related to #13264 ?

greggjaskiewicz, 2024-04-01:
quite possibly, although that issue was for linux

luzpaz, 2024-08-29:
Darn, forgot about this one.

luzpaz, 2024-10-31:
@hyarion any interest in this ?

hyarion, 2024-10-31:
@luzpaz maybe! :D
It is a good task to save for a new dev though. I think we can drop macOS tag on it as it would be useful on other systems as well.

I would recommend calling it something like "Reveal in Finder" on mac and "Open File Location" for other systems.

PR #19805 (2025-02-24): Gui: "Open File Location/Reveal in Finder" in Tree view
Fixes #12801 

~TODO:~

- [x] Handle windows and macOS 
- [x] Include screenshot of working version here

1. Right click on model and select option:

[image]

2. File opens up in explorer:

[image]