FreeCAD/FreeCAD #12802: [Problem] Array elements and their number are not taken into account in BIM managers

kaiwas, 2024-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you multiply any element using an array (and this can happen in many tasks), then the number of these elements does not change in any way in BIM managers.
You can see this situation in the pictures.

I tried changing the array parameters and it didn't help at all.

[image]
[image]

I would also like there to be a grouping of elements with similar parameters. Seeing 4/10/30 columns (or something else) with the same parameters in the list is not very useful. We have to recalculate them manually. But to see that there are 4/10/30 such columns would be very convenient.

[image]

### Full version info

```shell
any version / any system
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-07:
@yorikvanhavre is this BIM related or a function from Arch? I have little experience with Arch/BIM.

yorikvanhavre, 2024-03-08:
Indeed you can create arrays with Draft tools. However, the BIM quantities manager only looks at BIM objects, and the Draft array is not considered a BIM object, therefore it is not seen by the BIM manager, who only sees the base object.

This is a valid issue.

Only I'm not sure how to solve this. One solution that is easy to implement by the user is to wrap the array object in a BIM component, thus turning it into a BIM object. We have to check if the quantities are correctly set in that case, though. Would you test that @kaiwas ?

Another possibility would be to consider the Draft array as a kind of "feature" of its underlying BIM object and automatically "expand" the base object, both in quantities and IFC export. There is a partial support for that in IFC export already. The problem here is that IFC does not support the concept of arrays, so arrays of objects are exported as a series of individual objects. 

Maybe that's what we need to do here:

* In the quantities manager, detect if an object is part of a Draft array
* If yes, duplicate the entry the correct number of times
* Verify if the array is exported to IFC as a series of individual objects
* Verify if each individual object has its quantities set

maxwxyz, 2024-03-08:
Is it ok to leave the issue here or should it be in the BIM repo?

kaiwas, 2024-03-08:
> Only I'm not sure how to solve this. One solution that is easy to implement by the user is to wrap the array object in a BIM component, thus turning it into a BIM object. We have to check if the quantities are correctly set in that case, though. Would you test that @kaiwas ?

I'll try your recipe. But I admit that I don’t yet know how the IFS works with all its properties. I'm just looking for options for quick calculations, but BIM modeling itself is not familiar to me. But I'll try.

Another possible option would be to “destruct” the array. Those. Each element of the array would be transformed into a unique object when a button was pressed. Unfortunately, this method is not very flexible, because if you need to change the number of elements, you will have to re-create the array. And delete previously made “objects”.

I understand that taking into account massive numbers is not an easy task. Because the same “element” can be used in several different arrays or even be a separate object.
Here we clearly need to implement some additional logic. The first thing that comes to mind is that an object participating in arrays should not be automatically taken into account in quantities. But if you enable some kind of “checkbox”, it becomes visible for calculation.

kaiwas, 2024-03-08:
I'm not a programmer and don't know the deep capabilities. But I will try to think about the very logic of this feature.
This is very relevant because the fastest way to arrange an object with a certain step is an array.
(indication of difficulties via the link. Please read)
https://github.com/FreeCAD/FreeCAD/issues/12500

If there was another faster way, then the array would not be needed so often. Just imagine how many actions you need to do without an array in order to arrange the elements of the rafter system in the picture

kaiwas, 2024-03-10:
I don’t quite understand what I’m doing, but I tried to assign IFS properties to the array. The array appeared in the list as a single element without additional parameters.
It seems it's not easy.
[image]

yorikvanhavre, 2024-03-12:
Wtih 4133053 arrays are now correctly exported as an IfcElementAssembly, and contain a number of subelements. Next step is to verify the quantities.

closed 2024-03-18 by sliptonic