FreeCAD/FreeCAD #12805: [Problem] FEM ConstraintForce may generate values in an incorrect format causing a CalculiX error

FEA-eng, 2024-03-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While helping a user ([forum thread](https://forum.freecad.org/viewtopic.php?t=85921)), I've noticed that FEM ConstraintForce may generate values which can't be correctly read by CalculiX. The following error appears when submitting an analysis:

```
 *ERROR reading *CLOAD. Card image:
        6833,1,1.7302535485473438E-06
```

This is caused by the fact that CalculiX uses the F20.0 Fortran format to read the values and can only read 20 characters so the following value:

`1.7302535485473438E-06`

is read as:

`1.7302535485473438E-`

which is obviously incorrect.

Thus, it would be necessary to ensure that FEM ConstraintForce doesn't generate so many decimals and follows the F20.0 format required by CalculiX.

Model (just change the extension to .FCStd and regenerate the mesh, then try running this): 
[Number format error.zip](https://github.com/FreeCAD/FreeCAD/files/14535989/Number.format.error.zip)

@marioalexis84 Do you know how to fix it ?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-03-13:
> Model (just change the extension to .FCStd and regenerate the mesh, then try running this):
[Number format error.zip](https://github.com/FreeCAD/FreeCAD/files/14535989/Number.format.error.zip)

For me, gmsh fails to mesh the compound shape.

FEA-eng, 2024-03-13:
> For me, gmsh fails to mesh the compound shape.

Strange. But maybe you have different Gmsh settings. Mine are:

[image]

Especially the `Second Order Linear` property has to be true.

I only get a warning (2 ill-shaped tets) but no errors and the analysis works if I replace the force with pressure.

commit 5264d6cc (2024-03-13): Merge pull request #12932 from marioalexis84/fem-ccx_force_format

Fem: Format force constraint in CalculiX writer - fixes #12805
commit 7a07a613 (2024-03-13): Fem: Format force constraint in CalculiX writer - fixes #12805
closed 2024-03-13 by chennes