FreeCAD/FreeCAD #12814: [Feature request] UI/UX: tabbed panels

pierreporte, 2024-03-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following discussions with the Design Working Group (DWG) and @sliptonic, the described feature has been considered very useful: tabbed panels, or basically a column of tabs at the edge of the window, that expand into panels when selected while remaining discreet when unneeded. It is a feature that already exist in Solidworks as you can see in the screenshot below. Originally, it was inspired by issue #11072 as an alternative to the Properties panel window.

The tabbed panels could be used for a wide range of features, even replacing whole workbenches. It would be most useful as some sort of API to allow addons to create their own one.

This is a list of potential uses, most already discussed in the DWG:

* Tasks
* Properties (#11072)
* The material worbench @davesrocketshop 
* Appearance
* Fasteners, or a more general part library
* The insert component panel for assemblies @PaddleStroke 
* The new measuring toolsdd
* Help (short and contextual)
* Configurations
* Blocks (for TechDraw) @WandererFan
* [Ondsel Lens](https://github.com/Ondsel-Development/Ondsel-Lens)
* ...

There can be more so make your own list. There can be less so don’t hesitate to tell what shouldn’t be a tabbed panel at all.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Anything else?

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-08:
@pieterhijma I just saw your Ondsel video. You may be interested in this for configurations.

pieterhijma, 2024-03-19:
> @pieterhijma I just saw your Ondsel video. You may be interested in this for configurations.

Great, thanks for watching. I don't see an immediate connection, could you share what you have in mind? Regarding the feature request. Looks useful to me.

pierreporte, 2024-03-19:
@pieterhijma Instead of using the property views, a form would be generated in a panel. For a simple box, it could look like this:

[image]

To give another Solidworks example, it proposes a list of configuration, either already created or user-defined. You can right-click on the default configuration to create a new one, which adds to the list, ore chose an already existing one. It has forms similar to the mockup as well.

[image]

pierreporte, 2024-04-22:
@maxwxyz for the DWG project.